rust-embedded/discovery #26: GDB print format

spamwax, 2017-01-04:
Wasn't sure where to ask this question! At the end of [registers chapters](https://japaric.github.io/discovery/07-registers/type-safe-manipulation.html), you are showing us how to print content of `gpioe` register block. I noticed that your version of output for `print *gpioe` nicely shows the `struct`'s fields in terms of Rust module's path:

```gdb
(gdb) print *gpioe
$2 = f3::peripheral::gpio::Gpio {
  moder: f3::peripheral::gpio::Moder {
    register: volatile_register::RW<u32> {
      register: 0x55550000
    }
```

However mine, just shows the field names and their value:

```gdb
(gdb) print *gpioe
$3 = {
  moder = {
    register = {
      register = 1431633920
    }
  },
  otyper = {
    register = {
      register = 0
    }
  },
```

I tried this both on OSX's gdb (7.10.1.20160923) and Debian's (7.11.1-2+9+b1) and did not get output similar to yours. Is there any setting or argument that needs to be passed to gdb to enable that?

japaric, 2017-01-04:
You need GDB 7.12 or newer to get printing like that.

In theory, you could get something similar with older GDB releases using Rust's pretty printers (`rustup` installs them) but I never got that to work. See rust-lang/rust#36605

spamwax, 2017-01-05:
Thanks for the answer.
Current version that `brew` installs is 7.10. I installed the latest from [arm's website] for OSX (10.11.6). However, upon setting the breakpoint on main `gdb` crashes! I guess I have to wait for Debian(x86) to update its package so I can test it on my old Linux box and see if crash is unique to OSX.

[arm's website]: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

closed 2017-03-06 by spamwax