rust-embedded/discovery #28: No uucp group on Fedora 24 (possibly other distros)

cwoodall, 2017-01-14:
https://japaric.github.io/discovery/03-setup/linux.html

Typically those things belong to dialout on my systems, but as far as I can tell there is no uucp group by default on my machine

cwoodall, 2017-04-04:
It appears that the uucp group was deprecated in Fedora at some point and you are supposed to use dialout for this particular purpose. I wrote instructions for the Boston Embedded Systems Meetup's Bring and Blink Workshop (heavily based on this book but with some contextual changes) for Fedora 23+ (https://embedded-boston.github.io/tutorial-bring-and-blink-rust/03-setup/linux.html)

I can put together a PR if you approve

r-marques, 2018-06-20:
Can confirm that the instructions from @cwoodall work on Fedora 28.