rust-embedded/discovery #32: On Fedora 24 itm.txt file location is not correct

cwoodall, 2017-01-14:
From https://japaric.github.io/discovery/06-hello-world/README.html The following instructions appear to be wrong

```
# *nix
$ cd /tmp

# Windows
$ cd %TEMP%

$ itmdump itm.txt
```

Instead you must run `itmdump itm.txt` in the directory you ran gdb in with your current instructions. I believe you can make them correct by changing the following:

```
(gdb) monitor tpiu config internal itm.txt uart off 8000000
```

To

```
(gdb) monitor tpiu config internal /tmp/itm.txt uart off 8000000
```

adamvoss, 2018-02-09:
Where are you running `openocd`?

From the page you linked:
>  This should be the same directory from where you are running OpenOCD. It's very important that both itmdump and openocd are running from the same directory!

johnthagen, 2018-07-21:
Can confirm the instructions work as documented (without @cwoodall's change) when `itmdump` and `openocd` are both run in the same directory (e.g. `/tmp`) on macOS.

That quote @adamvoss quoted, might be better made into something more obvious.

PR #96 (2018-08-03): Highlight the fact that openocd and itmdump must be run in the same d…
...irectory

Fixes #32
closed 2018-08-03 by japaric