rust-embedded/discovery #35: Linker error: led_roulette...0.o uses VFP register arguments, but ... does not

getreu, 2017-02-12:
I migrated my dev system from Debian 8 to Debian 9. I followed https://japaric.github.io/discovery/03-setup/linux.html to set up my toolchain which worked well in Debian 8. Now with Debian 9 I get the following error:  `led_roulette...0.o uses VFP register arguments, but ... does not` (see log below).

I found [this post](http://stackoverflow.com/questions/9753749/arm-compilation-error-vpf-registered-used-by-executable-not-object-file#9753955) explaining: 
> Your target triplet indicates that your compiler is configured for the hard-float ABI. This means that the libgcc library will also be hardfp. The error message indicates that at least part of your system is using soft-float ABI. 

Any idea how can I fix this?
Thank you in advance

```
Compiling led-roulette v0.1.0 (file:///tmp/05-led-roulette)
error: linking with `arm-none-eabi-gcc` failed: exit code: 1
  |
  = note: "arm-none-eabi-gcc" "-L" "/home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o" "-o" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60" "-Wl,--gc-sections" "-nodefaultlibs" "-L" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps" "-L" "/tmp/05-led-roulette/target/debug/deps" "-L" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/build/f3-cdfa7b9c1efa4255/out" "-L" "/home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib" "-Wl,-Bstatic" "-Wl,-Bdynamic" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libpg-ac6e51f0e61b6a31.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libf3-123e742e98e90c5b.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x_memory_map-cc3041989572365a.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libcompiler_builtins_snapshot-1a4209288c3e868c.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/librlibc-3706ac36a4368afe.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-f673cf33e5b39258.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-92f00f85b2478495.rlib" "/tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libref_slice-62a5a8ad65c5543f.rlib" "/home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-78cdc976288fbcf3.rlib"
  = note: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: error: /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o uses VFP register arguments, /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60 does not
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: error: /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libf3-123e742e98e90c5b.rlib(f3-123e742e98e90c5b.0.o) uses VFP register arguments, /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60 does not
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libf3-123e742e98e90c5b.rlib(f3-123e742e98e90c5b.0.o)
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: error: /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x_memory_map-cc3041989572365a.rlib(stm32f30x_memory_map-cc3041989572365a.0.o) uses VFP register arguments, /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60 does not
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x_memory_map-cc3041989572365a.rlib(stm32f30x_memory_map-cc3041989572365a.0.o)
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: error: /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/librlibc-3706ac36a4368afe.rlib(rlibc-3706ac36a4368afe.0.o) uses VFP register arguments, /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60 does not
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/librlibc-3706ac36a4368afe.rlib(rlibc-3706ac36a4368afe.0.o)
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: error: /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-f673cf33e5b39258.rlib(cortex_m-f673cf33e5b39258.0.o) uses VFP register arguments, /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60 does not
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-f673cf33e5b39258.rlib(cortex_m-f673cf33e5b39258.0.o)
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: error: /home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-78cdc976288fbcf3.rlib(core-78cdc976288fbcf3.0.o) uses VFP register arguments, /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60 does not
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-78cdc976288fbcf3.rlib(core-78cdc976288fbcf3.0.o)
          /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/crt0.o: In function `_start':
          /build/newlib-XAuz1P/newlib-2.4.0.20160527/build/arm-none-eabi/libgloss/arm/../../../../libgloss/arm/crt0.S:421: undefined reference to `atexit'
          /build/newlib-XAuz1P/newlib-2.4.0.20160527/build/arm-none-eabi/libgloss/arm/../../../../libgloss/arm/crt0.S:423: undefined reference to `__libc_init_array'
          /build/newlib-XAuz1P/newlib-2.4.0.20160527/build/arm-none-eabi/libgloss/arm/../../../../libgloss/arm/crt0.S:429: undefined reference to `exit'
          /build/newlib-XAuz1P/newlib-2.4.0.20160527/build/arm-none-eabi/libgloss/arm/../../../../libgloss/arm/crt0.S:429: undefined reference to `__libc_fini_array'
          /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o:(.ARM.exidx.text._ZN4core3mem7size_of17h74cc3cbcf81503d9E+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o:(.ARM.exidx.text._ZN4core3ptr31_$LT$impl$u20$$BP$mut$u20$T$GT$15wrapping_offset17h644baddd8c4db2ceE+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o:(.ARM.exidx.text._ZN4core3ptr33_$LT$impl$u20$$BP$const$u20$T$GT$6offset17hdd27b6245f827a4eE+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o:(.ARM.exidx.text._ZN4core3ptr33_$LT$impl$u20$$BP$const$u20$T$GT$7is_null17hc997f70270022e5bE+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o:(.ARM.exidx.text._ZN4core3ptr4null17h43d11c6995bde0abE+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          /tmp/05-led-roulette/target/thumbv7em-none-eabihf/debug/deps/led_roulette-132d63accba18f60.0.o:(.ARM.exidx.text._ZN4core5slice67_$LT$impl$u20$core..ops..Index$LT$I$GT$$u20$for$u20$$u5b$T$u5d$$GT$5index17hd85f1523675f69baE+0x0): more undefined references to `__aeabi_unwind_cpp_pr0' follow
          /home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-78cdc976288fbcf3.rlib(core-78cdc976288fbcf3.0.o): In function `core::panicking::panic_fmt::h0735e1d1e2768939':
          core.cgu-0.rs:(.text._ZN4core9panicking9panic_fmt17h0735e1d1e2768939E+0x24): undefined reference to `rust_begin_unwind'
[... cut ...]
          /home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-78cdc976288fbcf3.rlib(core-78cdc976288fbcf3.0.o):(.ARM.exidx.text._ZN4core3fmt9Formatter12pad_integral17h53007ef69e7b4667E+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          /home/getreu/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-78cdc976288fbcf3.rlib(core-78cdc976288fbcf3.0.o):(.ARM.exidx.text._ZN4core3fmt3num54_$LT$impl$u20$core..fmt..Display$u20$for$u20$usize$GT$3fmt17hc953ff2732709e0aE+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
          collect2: error: ld returned 1 exit status
```

japaric, 2017-02-13:
`-nostartfiles` appears to be missing from the linker call. Is `.cargo/config` set up correctly? (i.e. what are its contents?)

getreu, 2017-02-13:
Oh, yes! My sync script only copied hidden  `.git` directories. The `.cargo` directories got lost and I did not notice. Thank you for your help! I would not have found out by myself (before giving up).

I can confirm that the toolchain setup instructions in your "Discovery"-book work well on Debian 9 Stretch (and Debian 8 Jessie).

closed 2017-02-13 by getreu