rust-embedded/discovery #36: Can't set breakpoint at main!

spamwax, 2017-03-06:
Following the book instruction from [Debug it](https://japaric.github.io/discovery/05-led-roulette/debug-it.html), I can setup the gdb to talk to openocd and load the program correctly, but I am unable to set a breakpoint:

```gdb
Reading symbols from target/thumbv7em-none-eabihf/debug/led-roulette...done
(gdb) target remote :3333
Remote debugging using :3333
pg::init () at ~/discovery/05-led-roulette/pg/src/lib.rs:19
19      pub unsafe fn init() {
(gdb) load
Loading section .text, size 0xa00 lma 0x8000000
Start address 0x8000040, load size 2560
Transfer rate: 1 KB/sec, 2560 bytes/write.
(gdb) break main
Function "main" not defined.
Make breakpoint pending on future shared library load? (y or [n])
```

Of course continuing program causes an exception?!
```gdb
Breakpoint 1 (main) pending.
(gdb) cont
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
pg::exception_handler () at ~/discovery/05-led-roulette/pg/src/lib.rs:27
27          bkpt!();
(gdb)
```

I am on OS X and use ARM's gcc toolchain:
`GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 7.12.1.20170215-git`

japaric, 2017-03-06:
Sounds like the linker is dropping the main symbol at link time. Could you check with `objdump`?

```
# is `<main>` there?
$ arm-none-eabi-objdump -Cd target/thumbv7m-none-eabihf/debug/led-roulette | less
```

It's a bug with some gcc versions. It has been reported before (#34). You have to install / use a different GCC version. #34 mentions an alternative GCC repo at the end of its description.

spamwax, 2017-03-06:
I tried the mentioned `px4` brew tap in #34, its gdb crashes immediately.
The only way to get the gdb 7.12's support for rust on OS X seems to be using the linker from `5_4-2016q3` (ARM's official gcc) and then downloading/using gdb from `6-2017-q1-update`

fluxchief, 2017-03-14:
I'm having the same issue. How is the initialization done? It reaches pg::init, but this doesn't call the main function. The stack backtrace at pg::init also does not show anything beyond this and $sp seems to be at the start of the stack. How is main supposed to get called?

jamesmunns, 2017-03-24:
I ran into this option, and was able to correct this by adding a `KEEP` directive to the linker script.

In `memory.x`, change this:
```
    /* Entry point: the reset handler */
    __reset = .;
    *(.text.start);
```

To this:

```
    /* Entry point: the reset handler */
    KEEP(*(.text.start));
    __reset = .;
    *(.text.start);
```

Then do a `xargo clean` followed by a `xargo build ...`

kcking, 2017-04-03:
the `px4/px4` tap fixed this issue for me

```$ brew info px4/px4/gcc-arm-none-eabi
px4/px4/gcc-arm-none-eabi: stable 20160928
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
/usr/local/Cellar/gcc-arm-none-eabi/20160928 (5,502 files, 447.2MB) *
  Built from source on 2017-04-03 at 19:42:16
From: https://github.com/px4/homebrew-px4/blob/master/gcc-arm-none-eabi.rb```

japaric, 2017-04-04:
Turns out newer versions of gcc were dropping the entry point `_reset`, which is marked as so in the linker script. I have modified the linker script to force the linker to keep the symbol around and released a new version of the f3 crate (v0.3.1) with the fix.

Tentatively closing as fixed but let me know if you run into any other linker error.

closed 2017-04-04 by japaric

cwoodall, 2017-04-04:
@japaric Thanks for the support on this issue! We were having this problem at the Boston Embedded Systems Meetup's Bring and Blink and some people were very impressed with your responsiveness and helpfulness in resolving this issue. Thank you for this book and all of the work you have done! It is awesome

durka, 2018-09-03:
Hi, this came up again for me. I'm using GCC 5.4 on Debian 9.5. The `main` symbol *is* present but GDB can't find it for some reason. I was able to set a breakpoint by line number.

```sh
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ arm-none-eabi-objdump -Cd target/thumbv7em-none-eabihf/debug/led-roulette | rg main
08000188 <led_roulette::main>:
 8000192:	e7ff      	b.n	8000194 <led_roulette::main+0xc>
 8000194:	e7fe      	b.n	8000194 <led_roulette::main+0xc>
08000196 <main>:
 8000444:	f7ff fea7 	bl	8000196 <main>

$ arm-none-eabi-gdb target/thumbv7em-none-eabihf/debug/led-roulette
GNU gdb (7.12-6+9+b2) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from target/thumbv7em-none-eabihf/debug/led-roulette...done.
(gdb) target remote :3333
Remote debugging using :3333
Reset () at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.5.3/src/lib.rs:491
491	    let pre_init: unsafe extern "C" fn() = __pre_init;
(gdb) load
Loading section .vector_table, size 0x188 lma 0x8000000
Loading section .text, size 0x3f0 lma 0x8000188
Loading section .rodata, size 0x12c lma 0x8000580
Start address 0x80003e4, load size 1700
Transfer rate: 708 bytes/sec, 566 bytes/write.
(gdb) break led_roulette::main
Function "led_roulette::main" not defined.
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb) break main.rs:11
Breakpoint 1 at 0x800018c: file src/05-led-roulette/src/main.rs, line 11.
(gdb) continue
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, led_roulette::main::h1bb7a04b9156fd55 () at src/05-led-roulette/src/main.rs:13
13	    let x = 42;
(gdb) 
```