rust-embedded/discovery #41: First example will not build.

Grazfather, 2017-09-17:
```bash
[grazfather ~/code/discovery/src/05-led-roulette]$ rustup default nightly
info: using existing install for 'nightly-x86_64-apple-darwin'
info: default toolchain set to 'nightly-x86_64-apple-darwin'

  nightly-x86_64-apple-darwin unchanged - rustc 1.22.0-nightly (277476c4f 2017-09-16)

[grazfather ~/code/discovery/src/05-led-roulette]$ xargo build --target thumbv7em-none-eabihf
   Compiling f3 v0.3.1
   Compiling cortex-m v0.1.6
error: `<core::cell::UnsafeCell<T>>::new` is not yet stable as a const fn
  --> /Users/graziano/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.1.6/src/interrupt.rs:13:24
   |
13 |         Mutex { inner: UnsafeCell::new(value) }
   |                        ^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: in Nightly builds, add `#![feature(const_unsafe_cell_new)]` to the crate attributes to enable

error: aborting due to previous error

error: Could not compile `cortex-m`.

To learn more, run the command again with --verbose.
```

ALPHA-60, 2017-09-21:
@Grazfather I managed to build this example by adding `#![feature(const_unsafe_cell_new)]` to `~/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.1.6/src/lib.rs`

japaric, 2017-09-21:
Thanks for the bug report, @Grazfather.

This error has been fixed in cortex-m v0.1.7. You'll probably need to remove the Cargo.lock file, which is next to the Cargo.toml file, to make this work. `cargo update` should update the lockfile as well.

@ALPHA-60 You are not supposed to modify the registry that comes crates.io; it's supposed to be immutable.

closed 2017-09-21 by japaric

Grazfather, 2017-09-21:
Confirmed it works now :)