rust-embedded/discovery #52: Compiling Debug builds fails with `undefined reference to 'rust_begin_unwind'`.

cldershem, 2017-10-09:
I have been unable to get any of examples from Chapter 06, Hello World, on to compile in debug without modifying `Cargo.toml`. 

```sh
$ xargo build --target thumbv7em-none-eabihf
Compiling usart v0.1.0 (file:///home/cldershem/Dev/LearningStuffs/LearnRust/DiscoveryBook/discovery/src/06-hello-world)
error: linking with `arm-none-eabi-gcc` failed: exit code: 1
  |
  = note: "arm-none-eabi-gcc" "-L" "/home/cldershem/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib"
  //
  // lots of stuff omitted
  //
  = note: /home/cldershem/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-1e711822e5d5d922.rlib(core-1e711822e5d5d922.0.o): In function `co
    re::panicking::panic_fmt::hb7c1829b06bc1b3c':
          core.cgu-0.rs:(.text.cold._ZN4core9panicking9panic_fmt17hb7c1829b06bc1b3cE+0x24): undefined reference to `rust_begin_unwind'
          collect2: error: ld returned 1 exit status
error: aborting due to previous error
error: Could not compile `hello-world`.
To learn more, run the command again with --verbose.
```
```sh
$ uname -a && xargo --version && rustc --version
Linux cldershem-xps 4.10.0-35-generic #39-Ubuntu SMP Wed Sep 13 07:46:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
xargo 0.3.9
cargo 0.23.0-nightly (e447ac7e9 2017-09-27)
rustc 1.22.0-nightly (417c73891 2017-10-05)
```

Compiling with `--release` works fine. I can get it to compile in `debug` if I change the default `opt-level` in the `Cargo.toml`.

```toml
[profile.dev]
opt-level = 1
```

I cannot be sure if this is something specific to my setup or if it's an issue with a newer nightly. Any guidance on how to troubleshoot this issue will be appreciated.

japaric, 2017-10-09:
This is a nightly regression (cf. rust-lang/rust#44853). You can workaround the issue by adding these two lines to Cargo.toml:

``` toml
[profile.dev]
codegen-units = 1
```

This workaround was applied to all Cargo projects in this repo some days ago; you probably have an old checkout.

cldershem, 2017-10-09:
I love that this is a fast passed enough world that "some days ago" is "an old checkout".

I did a pull and tested one of the chapters. Everything compiled without errors. Thanks for pointing that out, I should have done a pull first.

closed 2017-10-09 by cldershem