rust-embedded/discovery #61: `itmdump` is hard to find

kper, 2018-01-20:
It was hard to find out that `itmdump` is hosted on `crates.io` and I have to install it with `cargo install itm`. You should definitely point that out more explicitly!

adamvoss, 2018-02-08:
Do you have a suggestion?

It is [mentioned in chapter 6 when you first need it](https://github.com/japaric/discovery/blame/master/src/06-hello-world/README.md#L62-L65) in case you missed the [installation instructions in chapter 3](https://japaric.github.io/discovery/03-setup/README.html#itmdump) (which I did).

siobhanjacobson, 2020-06-15:
The details for installing `itmdump` are included now in the setup section: https://docs.rust-embedded.org/discovery/03-setup/index.html

closed 2020-06-15 by adamgreig