rust-embedded/discovery #66: Target directory not clear in 05-led-roulette/build-it

WimbledonLabs, 2018-02-28:
When executing the following from the `discovery/src/05-led-roulette` directory:

```xargo build --target thumbv7em-none-eabihf```

It's not clear the the resulting target is `discovery/target` since the previous page says "Now, jump into the `src/05-led-roulette directory`."

japaric, 2018-02-28:
> It's not clear the the resulting target is discovery/target

Ah yes, this changed due to the last minute change of moving to workspaces. We could workaround the issue by commiting symlinks that point to discovery/target. Though, I'm unsure if that would work on Windows (probably not).

WimbledonLabs, 2018-02-28:
Thanks for this tutorial! I was excited to see that you updated it. Things have been going well so far on Ubuntu 16.04 (minus some gdb weirdness that I need to look into).

commit 1770c191 (2018-02-28): add symlinks to /target

moving to workspaces means that now the target directory will now be at the root of the repo. This
doesn't match the text which says that target is right next to the Cargo.toml of each Cargo project.
Workaround (only on *nix?) the problem by adding symlinks

closes #66
commit e1c75ad7 (2018-02-28): Merge #67

67: add symlinks to /target r=japaric a=japaric

moving to workspaces means that now the target directory will now be at the root of the repo. This
doesn't match the text which says that target is right next to the Cargo.toml of each Cargo project.
Workaround (only on *nix?) the problem by adding symlinks

closes #66

cc @WimbledonLabs thoughts?
PR #67 (2018-02-28): add symlinks to /target
moving to workspaces means that now the target directory will now be at the root of the repo. This
doesn't match the text which says that target is right next to the Cargo.toml of each Cargo project.
Workaround (only on *nix?) the problem by adding symlinks

closes #66

cc @WimbledonLabs thoughts?
closed 2018-02-28 by bors[bot]