rust-embedded/discovery #71: Windows 10 can't clone repository: Aux is an invalid directory

BenArtes, 2018-03-03:
Along with the [long list](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365247(v=vs.85).aspx) of names that can't be used for files on windows, Aux breaks `git clone` on Windows 10, and I assume all other versions of Windows.

[Reference.](https://social.technet.microsoft.com/Forums/Azure/en-US/d1cf7394-a39b-46bf-823d-46f33b2f78d0/unable-to-create-folder-named-aux-anywhere-in-the-disk?forum=w7itprogeneral)

BenArtes, 2018-03-03:
So I setup a virtual ubuntu machine so I could rename all the */aux/* directories to auxiliary then cloned on windows.  I updated the .toml paths and couldn't get Chapter 05 example to build.  I'm still new to rust, though not to embedded.

So I'll have to go back and brush up on rust before I can push this further and setup a pull request.

BenArtes, 2018-03-04:
I revisited the issue today and noticed that I had only changed Chapter 5's TOML.  The Workspace has dependencies for other chapters meaning I had to change their TOMLs as well before I could correctly compile the examples.

I've made changes and uploaded them to [my fork](https://github.com/BenArtes/discovery).

If you'd like me to make a pull request, let me know.

japaric, 2018-03-10:
> If you'd like me to make a pull request, let me know.

Yes, please. That would be great.

BenArtes, 2018-03-12:
I'll squash / clean up the commits tonight and send over the pull request.  Which branch do you want me to send the PR to?

dantho, 2018-03-19:
Looking forward to the PR, @BenArtes , 
(In the meantime, I just cloned your Fork, and am starting to use it.  So far, so good.)

BenArtes, 2018-03-19:
Sorry, I should've posted here when I posted the PR.

PR was https://github.com/japaric/discovery/pull/72. And is in the staging branch now.

japaric, 2018-03-20:
@BenArtes Oh, you sent the PR against the staging branch; I didn't notice. You should have sent it against the master branch. The staging branch is a temporary branch used to run tests only; it gets overwritten by the test system on each tested PR.

japaric, 2018-03-20:
Fixed in #73

PR #73 (2018-03-20): Rename aux to auxiliary for Windows and update .toml files.
closes #71
closed 2018-03-20 by japaric