rust-embedded/discovery #76: Mention USART using ST-LINK (without separate USB-UART module)

jturner314, 2018-04-07:
On PCB version C and newer of the STM32F3DISCOVERY board, the embedded ST-LINK has virtual COM port support. This means that a separate USB-UART module (such as the CH340G module mentioned in the book) is not necessary with recent boards. This is really nice because you don't need to buy the extra hardware, and you can use one less USB port.

While it may be a good idea to continue using a separate USB-UART module in the book to ensure maximum compatibility across different STM32F3DISCOVERY PCB versions, it's probably worth mentioning that this isn't necessary on recent boards that have the newer ST-LINK.

Some observations about the virtual COM port:

* The "VCP configuration" section of the STM32F3DISCOVERY user manual says that the USART pins connected to the ST-LINK are PA9 and PA10, but that's incorrect. The correct pins are PC4 and PC5 (as shown in the electrical schematic, by inspecting the board itself, and with testing).

* On my Linux machine, the serial port is connected as `/dev/ttyACM0`.

* I see a lot less noise in ADC measurements when I use the USART support in the ST-LINK than when I use a separate USB-UART module. This could be a result of my cheap USB-UART module, but it may also be true for higher-quality modules.

* The solder bridges for connecting the USART pins to the ST-LINK are SB13 and SB15. They were already connected on my board, and the user manual says that they should be connected for PCB version C and newer, but it's probably worth suggesting that the user check that the bridges are connected. (In fact, this is might be a good way to determine the PCB version you have, because these solder bridges were not connected on PCBs before version C.)

By the way, many thanks for writing the Discovery book! It's what got me started with embedded Rust development.

JoshMcguigan, 2018-06-29:
Thanks for posting this. I did not purchase the CH340G module, but I do have version C of the STM32F3DISCOVERY board (at least I can confirm SB13 and SB15 are connected). 

Is anything else required to enable the virtual COM port support on the board? I'm on OSX, and I don't think I am able to see the port. The output of `ls /dev/cu.*` is listed below. I've tried testing as described in chapter 11.1 (https://japaric.github.io/discovery/11-usart/send-a-single-byte.html) using both of the devices listed below, and in both cases I don't see any activity on my minicom screen after launching the program.

```
$ ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port	/dev/cu.SOC
/dev/cu.MALS			/dev/cu.usbmodem14513
```

Do you know of any other setup or troubleshooting steps I can try?

jturner314, 2018-06-29:
I forgot to mention that I've been able to get this to work only if I've started `openocd` while the device is connected. It seems to work if I start `openocd` and then access the serial port; or start `openocd`, kill `openocd`, and then try to access the serial port. So `openocd` doesn't have to be running in my experience, but it has to run at least once while the device is connected (maybe to initialize the ST-LINK or something). By the way, the `openocd` command I'm using is `openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg`.

The devices that show up when I plug the STM32F3DISCOVERY board into my Linux laptop via USB are the following: `/dev/serial`, `/dev/sdc`, `/dev/sg2`, and `/dev/ttyACM0`. The one that's worked for serial communication is `/dev/ttyACM0`. Unfortunately, I don't have access to a Mac to test with.

I'd recommend the following. (You've probably already done most of these things, but just in case, I'll mention them.)

1. Make sure you're attaching the USB cable to the ST-LINK USB port on the STM32F3DISCOVERY, not the USER USB port.
2. Does OS X allow you to observe the kernel logs/messages? If so, make sure that it's recognizing the STM32F3DISCOVERY board. The kernel logs/messages may even tell you what device file the kernel has created for the serial port. You could also try [OS X's equivalent of `lsusb`](https://stackoverflow.com/questions/17058134/is-there-an-equivalent-of-lsusb-for-os-x) to make sure that the board is recognized.
3. Observe which devices appear when you plug the STM32F3DISCOVERY into your computer for the first time after it's been restarted. (Restart your computer, save the list of devices with `ls /dev > ~/devices_before`, plug in the STM32F3DISCOVERY, then save the new list of devices with `ls /dev > ~/devices_after`, then compare with `diff ~/devices_before ~/devices_after`.) I would assume that one of the new devices would be the serial device.
4. Make sure you've [configured `minicom` according to the book](https://japaric.github.io/discovery/10-serial-communication/nix-tooling.html#minicom).
5. Start `openocd` with `openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg` and make sure the LED on the STM32F3DISCOVERY starts changing colors.
6. Try connecting to the board with `minicom` again (while `openocd` is running).
7. Restart your computer and try again.

You could also try stepping through the test program with `gdb` to possibly get a better idea of what's happening.

JoshMcguigan, 2018-06-30:
```
$ diff disconnected.txt connected.txt 
17a18
> cu.usbmodem14513
25a27
> disk2
180a183
> rdisk2
189a193
> tty.usbmodem14513
```

`disk2` and `rdisk2` both return permission denied when trying to connect with minicom. `cu.usbmodem14513` and `tty.usbmodem14513` don't show any data in minicom after running the test program with `cargo run` and then `continue`. I am connecting through a USB-C to USB-A converter, so I'm not sure if that could be complicating things.

jturner314, 2018-07-02:
Based on a couple of articles talking about the ST-LINK "Virtual COM" feature [[1]](http://www.radioshuttle.de/en/nucleo-st-link-interface-2/) [[2]](https://www.carminenoviello.com/2015/03/02/how-to-use-stm32-nucleo-serial-port/), the device name on a Mac would be `/dev/tty.usbmodem*`, so it's probably the `/dev/tty.usbmodem14513` device for your hardware. It's worth trying the `/dev/cu.usbmodem14513` device, though, if the other one doesn't work.

I just realized the most probable reason why the chapter 11 example isn't working for you, though. As I mentioned in my first comment on this issue, the pins on the STM32F3 microcontroller that are connected to the ST-LINK for USART are pins PC4 and PC5. However, the pins used in the chapter 11 example of the book are PA9 and PA10. So, you need to change the example to use the pins connected to the ST-LINK. [I think this will work.](https://github.com/jturner314/discovery/commit/425a8e9f8a3ef12b28417bf98336b24185fa0521) The other examples in the book using USART need to be changed too in order to use the pins connected to the ST-LINK.

Don't forget to start `openocd` as I described in my previous comment, or the serial port won't work. (I've done that a few times.)

JoshMcguigan, 2018-07-02:
Switching out the pins as shown in [your fork](https://github.com/jturner314/discovery/commit/425a8e9f8a3ef12b28417bf98336b24185fa0521) worked! 

Thanks for the help. I think this information would be a great addition to the Discovery book.

jturner314, 2018-07-02:
Yay! Which device file ended up working? (Was it `/dev/tty.usbmodem14513`?)

JoshMcguigan, 2018-07-03:
Yes, it was `/dev/tty.usbmodem14513`

JoshMcguigan, 2018-07-16:
By the way, I just noticed that the latest version of the user manual has corrected the pinouts to PC4 & PC5 as you listed above. The revision history says it was fixed 5 days after your original post. 

https://www.st.com/content/ccc/resource/technical/document/user_manual/8a/56/97/63/8d/56/41/73/DM00063382.pdf/files/DM00063382.pdf/jcr:content/translations/en.DM00063382.pdf

japaric, 2018-11-27:
For those watching this issue. I have put up a PR (#148) to update the text and recommend using the Virtual COM.

therealprof, 2020-04-08:
@japaric It was just noticed that the PR is not merged due to the removal of the image which we'd like to keep. Any chance of getting that addressed?

closed 2020-04-26 by bors[bot]

7sDream, 2020-09-06:
Hello @jturner314 and @JoshMcguigan,

Sorry for creat new post in a closed issue. I just meet some trouble when read the serial communication section. And after some search I found this issue, which have some discussions on PCB version macOS enviroment, which close to my problem. So I took the liberty to ask some questions here.

I just bought a board of  PCB version D-01:

[image]

[image]

But when I plug it in my MacBook Pro labtop, I can't find any device name start with `tty.`(also tried after running `openocd`):

[image]

I notice SB13 and SB15 bridge of my board is little strange. It looks like there is solder on it, but the break of `<` shape still remain on it

[image]

Does this mean I bought a fake? Or is this actually a frequent occurrence, I only need to re-soldering by myself?

Thank you!

jturner314, 2020-09-06:
I don't have my board with me, but from memory, it looks very similar to your pictures, so I doubt yours is a fake. Maybe it's a manufacturing defect? According to the docs, SB13 and SB15 should be connected on PCB versions C and later. If you want to use the ST-LINK USART functionality, you'll need to make those connections somehow. I'd suggest just soldering the bridges yourself (maybe using a 0Ω SMD resistor or a short segment of stripped wire to help the solder bridge the gap). Alternatively, you could solder header pins to the pads labeled TX and RX, and then use jumpers to connect TX to PC5 and RX to PC4.

7sDream, 2020-09-07:
@jturner314 

OK, I will try to solder those bridge on next weekend and see if I can find the virtual COM device. 

Thank you for your information and suggestions.

sompo-cyber, 2026-09-09:
THANK SO MUCH SIR

commit 8c31c3a3 (2020-04-26): recommend using Virtual COM instead of an external USB <-> Serial module

this keeps the text about using the Serial module

closes #76