rust-embedded/discovery #80: Undefined reference to '__aeabi_memcpy'

damcio, 2018-04-12:
Hi,

I've been working my way through the tutorial lately but today I've hit a bump. First stm32f30x stopped building, because it wouldn't take latest (0.4.0) cortex-m-rt but insisted on previous version. I cloned it and changed Cargo.toml in both discovery and stm32f30x, added a dupe panic_fmt, and then when I tried to build using local dependency everything broke at last linking step, mainly getting errors as such:

`  = note: /home/damck/mgr/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-3ff1d9f0667fc7b0.rlib(cortex_m-3ff1d9f0667fc7b0.cortex_m0.rcgu.o): In function `core::ptr::swap_nonoverlapping_bytes':
          /home/damck/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/libcore/ptr.rs:223: undefined reference to `__aeabi_memcpy'
`

And I can't seem to find what else I could do. Is this the 'outdated' part I've noticed on the repository? Any clue what else I might be doing wrong/missing?

Sorry for a newb issue, but I'd rather use this than mail.

japaric, 2018-04-12:
> it wouldn't take latest (0.4.0) cortex-m-rt

Did you hit some problem with v0.3.x that prompted you to try v0.4.0?

This repo hasn't been updated to v0.4.0 of cortex-m-rt given that it's been out for less than a week. It's not a trivial change to do the upgrade as v0.4.0 and v0.3.x are incompatible (hence the different minor version).

I would suggest continue using v0.3.x of cortex-m-rt, which is the version [we are testing](https://travis-ci.org/japaric/discovery).

damcio, 2018-04-13:
Yeah, when stm32f30x tried to build using 0.3.15 it'd stop at 
`
extern crate compiler_builtins
`
being missing. Is the solution to just use earlier rustc that didn't remove those yet or is issue elsewhere? I noticed that cortex-m-rt and some other repositories adapted towards getting rid of the crate, but I couldn't really find the source of changes.

closed 2018-04-13 by damcio