rust-embedded/discovery #85: Build is broken due to panic_fmt being broken

MaxStrange, 2018-06-08:
Trying to compile the LED example from chapter 5 results in this:

```
$> cargo build --target thumbv7em-none-eabihf
   Compiling cortex-m-rt v0.3.15
error[E0522]: definition of an unknown language item: `panic_fmt`
 --> C:\Users\maxst\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.15\src\lang_items.rs:3:1
  |
3 | #[lang = "panic_fmt"]
  | ^^^^^^^^^^^^^^^^^^^^^ definition of unknown language item `panic_fmt`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0522`.
error: Could not compile `cortex-m-rt`.

To learn more, run the command again with --verbose.
```

The reason for this is due to this:
https://users.rust-lang.org/t/psa-breaking-change-panic-fmt-language-item-removed-in-favor-of-panic-implementation/17875

dholth, 2018-06-15:
Which nightly should we install so that it still works?

MaxStrange, 2018-06-15:
I've been using nightly-2018-05-20, but I'd guess most nightlies that are around the latest code commits in this repo would work.

dholth, 2018-06-15:
2018-06-01 works too.

AJAnderson, 2018-06-23:
How do we fix this? part of the problem is that the cargo.toml specifies an older version of cortex-m-rt, which you can fix naively by going to the source and adding #![feature(panic_implementation)], but looking at more recent versions of cortex-m-rt it looks totally different.

droogmic, 2018-06-23:
Rolling back versions is a short term solution which I would rather not do. Seeing as new versions of cortex-m-rt have removed the abort-on-panic feature, I thought I would try removing it, adding the panic-abort crate, and bumping all the dependencies (all in 05/auxiliary/Cargo.toml). It still doesn't build though, so not sure how to fix it:
```
error: language item required, but not found: `panic_impl`
error: language item required, but not found: `eh_personality`
```

(See: https://users.rust-lang.org/t/psa-breaking-change-panic-fmt-language-item-removed-in-favor-of-panic-implementation/17875)

eldruin, 2018-06-24:
This just bit me on chapter 16 as well.
I was able to get it to compile by adapting the panic implementation (see diff)
```diff
diff --git a/src/16-punch-o-meter/auxiliary/src/lib.rs b/src/16-punch-o-meter/auxiliary/src/lib.rs
index 5484cd7..e268ec8 100644
--- a/src/16-punch-o-meter/auxiliary/src/lib.rs
+++ b/src/16-punch-o-meter/auxiliary/src/lib.rs
@@ -2,6 +2,7 @@
 
 #![feature(lang_items)]
 #![feature(use_extern_macros)]
+#![feature(panic_implementation)]
 #![no_std]
 
 extern crate cortex_m;
@@ -49,20 +50,17 @@ pub fn init() -> (Lsm303dlhc, Delay, MonoTimer, ITM) {
     (lsm303dlhc, delay, mono_timer, cp.ITM)
 }
 
-#[lang = "panic_fmt"]
-unsafe extern "C" fn panic_fmt(
-    args: ::core::fmt::Arguments,
-    file: &'static str,
-    line: u32,
-    col: u32,
-) -> ! {
-    let itm = &mut *ITM::ptr();
-
-    itm::write_str(&mut itm.stim[0], "PANIC at '");
-    itm::write_fmt(&mut itm.stim[0], args);
-    iprintln!(&mut itm.stim[0], "', {}:{}:{}", file, line, col);
-
+use core::panic::PanicInfo;
+#[panic_implementation]
+fn panic(_info: &PanicInfo) -> ! {
+    unsafe {
+        let itm = &mut *ITM::ptr();
+
+        let location = _info.location().unwrap();
+        itm::write_str(&mut itm.stim[0], "PANIC at '");
+        itm::write_str(&mut itm.stim[0], _info.payload().downcast_ref::<&str>().unwrap());
+        iprintln!(&mut itm.stim[0], "', {}:{}:{}", location.file(), location.line(), location.column());
+    }
     cortex_m::asm::bkpt();
-
     loop {}
 }
```
However, I am getting a linking error now:
```
       Fresh semver-parser v0.7.0
       Fresh libc v0.2.42
       Fresh vcell v0.1.0
       Fresh bare-metal v0.1.2
       Fresh aligned v0.1.2
       Fresh nb v0.1.1
       Fresh void v1.0.2
       Fresh untagged-option v0.1.1
       Fresh r0 v0.2.2
       Fresh cast v0.2.2
       Fresh m v0.1.1
       Fresh semver v0.9.0
       Fresh time v0.1.40
       Fresh volatile-register v0.2.0
       Fresh embedded-hal v0.2.1
       Fresh num-traits v0.2.5
       Fresh rustc_version v0.2.2
       Fresh cortex-m v0.3.1
       Fresh cortex-m v0.4.3
       Fresh embedded-hal v0.1.3
[... 6 lines cut ...]
       Fresh stm32f30x-hal v0.1.2
       Fresh f3 v0.5.3
       Fresh aux16 v0.1.0 (file:///home/eldruin/projects/f3/discovery/src/16-punch-o-meter/auxiliary)
   Compiling punch-o-meter v0.1.0 (file:///home/eldruin/projects/f3/discovery/src/16-punch-o-meter)
     Running `rustc --crate-name punch_o_meter src/16-punch-o-meter/src/main.rs --crate-type bin --emit=dep-info,link -C codegen-units=1 -C debuginfo=2 -C metadata=7f24593758ba53f2 -C extra-filename=-7f24593758ba53f2 --out-dir /home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps --target thumbv7em-none-eabi -L dependency=/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps -L dependency=/home/eldruin/projects/f3/discovery/target/debug/deps --extern aux16=/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libaux16-e52008c08692899e.rlib --extern m=/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libm-c2d2fe2c5ae773a1.rlib -L /home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/build/f3-4fc456239fba8fde/out -L /home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/build/cortex-m-rt-b4b3f7fbab5c6b77/out`
error: linking with `arm-none-eabi-gcc` failed: exit code: 1
  |
  = note: "arm-none-eabi-gcc" "-L" "/home/eldruin/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabi/lib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/punch_o_meter-7f24593758ba53f2.punch_o_meter0.rcgu.o" "-o" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/punch_o_meter-7f24593758ba53f2" "-Wl,--gc-sections" "-nodefaultlibs" "-L" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps" "-L" "/home/eldruin/projects/f3/discovery/target/debug/deps" "-L" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/build/f3-4fc456239fba8fde/out" "-L" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/build/cortex-m-rt-b4b3f7fbab5c6b77/out" "-L" "/home/eldruin/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabi/lib" "-Wl,-Bstatic" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libm-c2d2fe2c5ae773a1.rlib" "-Wl,--start-group" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libaux16-e52008c08692899e.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libf3-5410013488dc6fe6.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libstm32f30x_hal-59cba9db1666eb02.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libstm32f30x-a042881db84c13ed.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libcortex_m_rt-94b764977b1bc8aa.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libr0-cb0bc673be809319.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libcortex_m-b64df0b1502144eb.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/liblsm303dlhc-f3ee7ad6c49eb30c.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libcast-17056d054cc49be3.rlib" "/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libl3gd20-0f7e94447fe7 [... cut ...] g/deps --extern aux16=/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libaux16-e52008c08692899e.rlib --extern m=/home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/deps/libm-c2d2fe2c5ae773a1.rlib -L /home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/build/f3-4fc456239fba8fde/out -L /home/eldruin/projects/f3/discovery/target/thumbv7em-none-eabi/debug/build/cortex-m-rt-b4b3f7fbab5c6b77/out` (exit code: 101)
```

Can anybody help me here?

japaric, 2018-06-28:
Sorry for the trouble and the delay. #86 has updated the whole book to work with the latest version of the cortex-m crates as well as with recent nightlies.

closed 2018-06-28 by japaric