rust-embedded/discovery #103: Simplify openocd launching using board configuration

johnthagen, 2018-08-05:
Once [OpenOCD issue 200](https://sourceforge.net/p/openocd/tickets/200/) is fixed, we can update the guide to simplify the launching of `openocd` from two arguments to one argument.

[Here](https://japaric.github.io/discovery/03-setup/verify.html#first-openocd-connection):

> `openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg`

Can become:

> `openocd -f board/stm32f3discovery.cfg`

But this can't be merged until OpenOCD fixes the issue linked above because the current `stm32f3discovery.cfg` points to the older `interface/stlink-v2-1.cfg`.

johnthagen, 2018-08-06:
Confirmed through discussions on the issue that this has been fixed in master and should then be fixed in the next OpenOCD release.

japaric, 2018-08-19:
Nice!

Before landing this change we should also check that easy to install the latest OpenOCD release on the 3 major OSes (specially on Windows).

therealprof, 2018-08-19:
@johnthagen One of my biggest beefs with OpenOCD is that they're extremely release-unhappy.

@japaric Yeah, that's trivial. For Windows there're several sites providing packages, e.g. http://gnutoolchains.com/arm-eabi/openocd/. For macOS there's homebrew: `brew install open-ocd`, for Linux there're a couple of different ways how to obtain it but it is always possible to build it yourself.

johnthagen, 2018-08-19:
And just in case anyone on Mac finds this thread and wants to try this out before the next OpenOCD release, the Homebrew command to get the latest master (with the unreleased fix) is:

```bash
$ brew install --HEAD openocd
```