rust-embedded/discovery #112: cargo size -- --version yields an error instead of the expected output

twilco, 2018-09-07:
Following the guide [here](https://rust-embedded.github.io/discovery/03-setup/README.html), running `cargo size -- --version` results in the following error:

```
dyld: Library not loaded: @rpath/libLLVM.dylib
  Referenced from: /Users/twilcock/.rustup/toolchains/nightly-x86_64-apple-darwin/bin/llvm-size
  Reason: image not found
```

I think there's been changes recently in how LLVM is referenced by the Rust compiler, so maybe that has something to do with it?  The other commands on that page worked without issue.

I'm using a 2014 Macbook Pro.  Here is my Rust version:

```
$ rustc --version                        
rustc 1.30.0-nightly (6e0f1cc15 2018-09-05)
```

twilco, 2018-09-07:
Ahh, probably related to this: #https://github.com/rust-lang/rust/pull/54018

japaric, 2018-09-15:
Thanks for the report.

I think this has been fixed upstream as of several days ago. We are also recommending nightly-2018-09-13, which doesn't have this problem, in #114 so I'm going to close this issue.

If you encounter any other problem with the tooling please let us know in a new issue!

closed 2018-09-15 by japaric