rust-embedded/discovery #113: fail to build 05-led-roulette

xieyuheng, 2018-09-08:
```
  130  /home/xyh/magismoke/discovery/src/05-led-roulette (master)
px cargo build --target thumbv7em-none-eabihf
[proxychains] config file found: /home/xyh/.proxychains/proxychains.conf
[proxychains] preloading /usr/lib/libproxychains4.so
    Updating registry `https://github.com/rust-lang/crates.io-index`
 Downloading cortex-m-rt v0.5.4
 Downloading r0 v0.2.2
 Downloading panic-abort v0.2.0
 Downloading cortex-m v0.5.7
 Downloading f3 v0.6.1
 Downloading aligned v0.2.0
 Downloading bare-metal v0.2.3
 Downloading volatile-register v0.2.0
 Downloading vcell v0.1.0
 Downloading stm32f30x-hal v0.2.0
 Downloading lsm303dlhc v0.2.0
 Downloading l3gd20 v0.2.0
 Downloading stm32f30x v0.7.1
 Downloading embedded-hal v0.2.1
 Downloading nb v0.1.1
 Downloading cortex-m-rt v0.6.1
 Downloading cortex-m-rt-macros v0.1.0
 Downloading generic-array v0.11.1
   Compiling proc-macro2 v0.4.18
   Compiling unicode-xid v0.1.0
   Compiling libc v0.2.43
   Compiling rand_core v0.2.1
   Compiling cortex-m-rt v0.6.1
   Compiling cortex-m v0.5.7
   Compiling typenum v1.10.0
   Compiling vcell v0.1.0
   Compiling stm32f30x v0.7.1
   Compiling bare-metal v0.2.3
   Compiling aligned v0.2.0
   Compiling r0 v0.2.2
   Compiling nb v0.1.1
   Compiling void v1.0.2
   Compiling f3 v0.6.1
   Compiling cortex-m-rt v0.5.4
   Compiling cast v0.2.2
   Compiling panic-abort v0.2.0
   Compiling rand v0.5.5
   Compiling volatile-register v0.2.0
   Compiling embedded-hal v0.2.1
   Compiling quote v0.6.8
   Compiling generic-array v0.11.1
   Compiling syn v0.14.9
   Compiling l3gd20 v0.2.0
   Compiling lsm303dlhc v0.2.0
   Compiling cortex-m-rt-macros v0.1.0
   Compiling stm32f30x-hal v0.2.0
   Compiling aux5 v0.1.0 (file:///home/xyh/magismoke/discovery/src/05-led-roulette/auxiliary)
   Compiling led-roulette v0.1.0 (file:///home/xyh/magismoke/discovery/src/05-led-roulette)
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "/home/xyh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/led_roulette-7003b953f9707fdd.2gvdue3lexsze7ea.rcgu.o" "-o" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/led_roulette-7003b953f9707fdd" "--gc-sections" "-L" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/xyh/magismoke/discovery/target/debug/deps" "-L" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-306f5924e0119627/out" "-L" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-107e2318a0bdb4e4/out" "-L" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/build/f3-aee5918ef6c0db4e/out" "-L" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/build/stm32f30x-fc5c0604003c8138/out" "-L" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-7a556696fd7220d1/out" "-L" "/home/xyh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libaux5-e6fd8fc7a8c5788e.rlib" "--start-group" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_abort-02324c692a31f947.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libf3-8b54b1d3345a8a66.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x_hal-c95c76d59f85822c.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-657d932de92cf175.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/liblsm303dlhc-197262740cce284e.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libcast-76f018da73c28584.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libl3gd20-5e8b472c3e812d01.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-2f978d6ba8eae7de.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libtypenum-70f6e206bc16c525.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-fa4b1f418deed0cf.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libvoid-89a68ad6553fb210.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libnb-fd72e2ca23440bb9.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-fe9359f76589f893.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libr0-91e7596db01c5ee4.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-a7bffe7c29cd199e.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-86534ccb05932bfd.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libvcell-f00ec07c8e9caa2f.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-4c7ecdb5fa6ab06f.rlib" "/home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libaligned-60940c5763543bdb.rlib" "/home/xyh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-b276b64f149a1ab9.rlib" "--end-group" "/home/xyh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-43525f2d3bc6e46d.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: undefined symbol: WWDG
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: undefined symbol: PVD
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: undefined symbol: TAMP_STAMP
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: undefined symbol: RTC_WKUP
[... 52 lines cut ...]
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: undefined symbol: DMA1_CH7
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: undefined symbol: ADC1_2
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: undefined symbol: USB_HP_CAN_TX
          >>> referenced by stm32f30x.8wanakcq-cgu.10
          >>>               stm32f30x-92f986590f4ce34a.stm32f30x.8wanakcq-cgu.10.rcgu.o:(__INTERRUPTS) in archive /home/xyh/magismoke/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-92f986590f4ce34a.rlib

          rust-lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)

error: aborting due to previous error

error: Could not compile `led-roulette`.

To learn more, run the command again with --verbose.

  101  /home/xyh/magismoke/discovery/src/05-led-roulette (master)
```

japaric, 2018-09-08:
Sorry about this, I published a new version of the stm32f30x crate without making adequate preparations  and testing, but the problem should now be fixed.

To get the fix you'll have to update some crates in your dependency graph by either deleting the Cargo.lock file or calling `cargo update`.

Let me know if this still doesn't work after you have tried that!

xieyuheng, 2018-09-09:
Thank you Sir.
The problem is fixed now.

---------------

In another note.
Is the use of `target -> ../../target/` safe with cargo ?

I understand it is used to reduce compiling time,
but if I `cargo build` in `05-led-roulette`
and then in `06-hello-world`,
and then again in `05-led-roulette`,
only the first build will success.

something like the following will happen :

```
  /home/xyh/magismoke/discovery/src/06-hello-world (master)
cargo build
   Compiling vcell v0.1.0
   Compiling aligned v0.2.0
   Compiling r0 v0.2.2
   Compiling bare-metal v0.2.3
   Compiling nb v0.1.1
   Compiling void v1.0.2
   Compiling cast v0.2.2
   Compiling cortex-m v0.5.7
   Compiling typenum v1.10.0
   Compiling cortex-m-rt v0.6.2
   Compiling stm32f30x v0.7.1
   Compiling volatile-register v0.2.0
   Compiling embedded-hal v0.2.1
   Compiling f3 v0.6.1
   Compiling cortex-m-rt v0.5.7
   Compiling generic-array v0.11.1
   Compiling lsm303dlhc v0.2.0
   Compiling l3gd20 v0.2.0
   Compiling stm32f30x-hal v0.2.0
   Compiling aux6 v0.1.0 (file:///home/xyh/magismoke/discovery/src/06-hello-world/auxiliary)
   Compiling hello-world v0.1.0 (file:///home/xyh/magismoke/discovery/src/06-hello-world)
error: language item required, but not found: `eh_personality`

error: aborting due to previous error

error: Could not compile `hello-world`.

To learn more, run the command again with --verbose.

  101  /home/xyh/magismoke/discovery/src/06-hello-world (master)
```

and

```
  /home/xyh/magismoke/discovery/src/05-led-roulette (master)
cargo build
   Compiling panic-abort v0.2.0
   Compiling aux5 v0.1.0 (file:///home/xyh/magismoke/discovery/src/05-led-roulette/auxiliary)
   Compiling led-roulette v0.1.0 (file:///home/xyh/magismoke/discovery/src/05-led-roulette)
error: language item required, but not found: `eh_personality`

error: aborting due to previous error

error: Could not compile `led-roulette`.

To learn more, run the command again with --verbose.

  101  /home/xyh/magismoke/discovery/src/05-led-roulette (master)
ll
```

xieyuheng, 2018-09-09:
and the `Cargo.lock` file follows this `target -> ../../target/` to `../../Cargo.lock`

japaric, 2018-09-09:
> Is the use of target -> ../../target/ safe with cargo ?

That's just a convenience and has no effect on the build. All the projects are part of a [workspace] so they share their build artifacts in the `target` directory that's in the root of the repository.

[workspace]: https://doc.rust-lang.org/book/second-edition/ch14-03-cargo-workspaces.html

> but if I cargo build in 05-led-roulette

If you read through the book in order it'll tell that you have to use `cargo build --target thumbv7em-none-eabihf` in projects 05 and 06. You can use just `cargo build` in the other projects because `thumbv7em-none-eabihf` has been set as default compilation target in `.cargo/config`.

``` console
$ cd 05-led-roulette
$ cargo clean
$ cargo build --target thumbv7em-none-eabihf && echo OK
    Finished dev [unoptimized + debuginfo] target(s) in 38.91s
OK

$ cd ../06-hello-world
$ cargo build --target thumbv7em-none-eabihf && echo OK
    Finished dev [unoptimized + debuginfo] target(s) in 0.30s
OK
```

xieyuheng, 2018-09-09:
Thanks again for your patient explainntaions.
I am closing this issue.

closed 2018-09-09 by xieyuheng