rust-embedded/discovery #117: sugestions for 5 roulette

DaQue, 2018-10-05:
First I'm a absolute beginner so I hope you understand these suggestions are to help other absolute beginners and probably seem obvious to a more advanced user and do not imply something is "broken". 

First it It could be made clear the the first build of the src/05-led-roulette/main.rs does not actually  make the actual roulette led spin as in the photo  when compiled and uploaded. Its something you work up to by the end of the chapter. I flashed the src/05-led-roulette/main.rs and spent a lot of time trying to figure out what I did wrong because I didn't see the roulette leds spinning output. I looked at the source and had no idea how it could make it do that. 

Second the warning "NOTE Be sure to compile this crate without optimizations" is 100% accurate but explaining  that the provided Config.toml file does not enable optimizations so don't change it would have been helpful. 

Next, I think this one may already be fixed in the next release by looking at the source to rust-embedded/discovery is I had to change (gdb) break led_roulette::main to (gdb) break led_main. 

A confidence building option for something to add before the roulette chapter  would be a super simple just build a supplied blink one led main.rs  file and flash and run it using a script if possible. Next have them change one variable for a delay (i.e blink much faster)  so you can tell you modified the program and everything still works.  This would handle making sure openOCD and gdb worked with out any of the details to be filled in later.

therealprof, 2018-10-05:
@DaQue These are good suggestions. Would you like to take a spin at pull requests addressing these shortcomings?

therealprof, 2018-10-05:
Do not be afraid, there's always a first time and I'll be happy to guide you.

DaQue, 2018-10-05:
I'll look up what's actually goes in a pill request and give it a go this weekend.  I've talked to you on iirc and may chat about it there.

DaQue, 2018-10-06:
Just checking.   It looks like the two pull requested I did for this got submitted ok. Let me know if not. I’ll review my original comments to see if I addressed all of them I could.

therealprof, 2018-10-06:
@DaQue I see only 1 PR (https://github.com/rust-embedded/discovery/pull/118) which already got merged.

DaQue, 2018-10-07:
ok I'm not sure what I did differently. I deleted my old fork repository and made a new one with some minor changes about optimizations.  

https://github.com/DaQue/discovery/edit/master/src/05-led-roulette/build-it.md

Do I "Commit directly to the master branch." or "Create a new branch for this commit and start a pull request." to get it up torust-embedded/discovery/? I thought I did both ways with my 2nd set of changes but neither seemed to go upstream.

therealprof, 2018-10-07:
No need to delete the forks, after a PR is merged you'll automatically get the options to delete the branch (in the PR) so you don't have outdated branches hanging around.