rust-embedded/discovery #123: Images aren't showing up in documentation

adamgreen, 2018-10-13:
For example, if I navigate to https://rust-embedded.github.io/bookshelf/discovery/02-requirements/index.html none of the images linked from this page are rendered.

An example of a broken link found on this page:
https://rust-embedded.github.io/assets/f3.jpg

I think it should actually be:
https://rust-embedded.github.io/bookshelf/discovery/assets/f3.jpg

adamgreen, 2018-10-13:
It appears that PR #121 is attempting to fix the image links part of this issue.

ChouzArt, 2018-10-18:
Didn't notice the 404 on those relative links. Could be more but already made a new PR to fix those too

adamgreen, 2018-10-28:
This is a duplicate of https://github.com/rust-embedded/bookshelf/issues/4

closed 2018-10-28 by adamgreen