rust-embedded/discovery #147: Empty binary when compiling for thumbv7em-none-eabi

Kixunil, 2018-11-27:
When compiling to `thumbv7em-none-eabi`, the binary is empty - no `.text` section, only debug symbols. This doesn't happen when compiling to `thumbv7em-none-eabihf`

tomboland, 2019-01-17:
I'm getting empty binaries compiling to thumbv7em-none-eabihf, as well as thumbv7em-none-eabi, on nightly as well as stable.  This is just with the cortex-m-quickstart example code, from following the rust embedded book.  

```
> cargo size --bin srsly --release  -- -A
    Finished release [optimized + debuginfo] target(s) in 0.16s
srsly  :
section             size   addr
.ARM.attributes       54    0x0
Total                 54
```
```
> cargo size --bin srsly  -- -A
    Finished dev [unoptimized + debuginfo] target(s) in 0.15s
srsly  :
section             size   addr
.ARM.attributes       54    0x0
Total       
```
This is on Windows 10, using the arm-none-eabi-ld as the linker.  Using the lld linker completely fails for me, on stable but not nightly, with:

```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "C:\\Users\\tom.boland\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.15ck1gqmguqxhuu6.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.2827aqqr83day0yn.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.2p6633go9k7vkrso.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.3fs8tsjtviqlz5y2.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.3hxq5o21kisybiiq.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.48ymk3echnpzf6dm.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.4sj2ckaxmdrydj4y.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.5286991edaqv77be.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.eh3l0hp6k36qvbi.rcgu.o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b.h5lypc3pnkftftj.rcgu.o" "-o" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\examples\\exception-f7b20df2fa11349b" "--gc-sections" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\debug\\deps" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\build\\srsly-7a880dcfc8c6131b\\out" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-05c503b0a2309d03\\out" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-rt-9bfe4635291c192e\\out" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-semihosting-4b40e0d53fdc1196\\out" "-L" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\build\\stm32f4-664ad3117ff0ecc8\\out" "-L" "C:\\Users\\tom.boland\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib" "-Bstatic" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcortex_m_semihosting-4540d7aa17e7115e.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcortex_m_rt-500d2346546ba1f1.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libr0-223cdb255103aafa.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcortex_m-bfcd5ea463d446a0.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libvolatile_register-e5f650978dfb9b50.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libvcell-199117581be49394.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libbare_metal-f7a252d944619522.rlib" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libaligned-e9cea8fdb6d1299b.rlib" "--start-group" "C:\\Users\\tom.boland\\nucleo-rust\\srsly\\target\\thumbv7em-none-eabihf\\debug\\deps\\libpanic_halt-864f8010aa796478.rlib" "C:\\Users\\tom.boland\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib\\libcore-879310dc3b96af61.rlib" "--end-group" "C:\\Users\\tom.boland\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib\\libcompiler_builtins-f49b1fd2880b372d.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
```

My cargo.toml
```
[package]
authors = ["tom.boland <tom@t0mb.net>"]
[... cut ...]
panic-halt = "0.2.0"

# Uncomment for the panic example.
# panic-itm = "0.4.0"

# Uncomment for the allocator example.
# alloc-cortex-m = "0.3.5"

[dependencies.stm32f4]
features = ["rt", "stm32f401"]
version = "0.6.0"

# this lets you use `cargo fix`!
[[bin]]
name = "srsly"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
```

Should I open a new issue?

japaric, 2019-02-08:
> Should I open a new issue?

Yes, please. This issue was caused by overriding the .cargo/config file and removing the extra -Tlink.x by mistake.

closed 2019-02-08 by japaric