rust-embedded/discovery #149: undefined debug reason 7 (stm32f303 Discovery)

polyc, 2018-11-29:
I followed discovery book up to the ["flash it"](https://rust-embedded.github.io/discovery/05-led-roulette/flash-it.html) section and, when i run `target remote :3333` on gdb with openocd running on another shell, openocd returns this output:

```
$ openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg

Open On-Chip Debugger 0.9.0 (2018-01-24-01:05)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v32 API v2 SWIM v22 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.888208
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x10036422
Info : flash size = 256kbytes
undefined debug reason 7 - target needs reset
```

And when i try to load led-roulette, it gives me this, as if it hasn't flashed anything on the board:

```
(gdb) load
Start address 0x0, load size 0
Transfer rate: 0 bits in <1 sec.
```

My environment is:

- Lubuntu 16.04 LTS
- rustc 1.31.0-beta.19 (42053f9f0 2018-11-26)
- itmdump 0.3.1
- LLVM version 8.0.0svn
  Optimized build.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: skylake
- openocd 0.9.0
- minicom 2.7
- arm-none-eabi-gdb 7.10

therealprof, 2018-11-29:
Sounds like a miscompiled binary (wrong load address). Can you post the output of `readelf -l` on the binary you're trying to flash.

polyc, 2018-11-29:
i totally missed this line when i ran cargo: ` # make sure you are in the `src/05-led-roulette` directory`.
Thanks for the help. Anyway, "undefined debug reason 7" it's still there when i launch `target remote :3333`, but now loads correctly the executable. Any problems with that?

therealprof, 2018-11-29:
That's okay, just a note that OpenOCD doesn't know which state the MCU is in at the moment.

polyc, 2018-11-29:
ok thank you.

closed 2018-11-29 by polyc