rust-embedded/discovery #167: Brew cask issue installing gcc-arm-embedded

kylestew, 2019-02-07:
So you don't waste time trying to follow the instructions in the book to no end. The maintainers at the caskroom repo pulled this cask citing that is should live in Brew. It's not there yet. Head over to the ticket for some alternative actions. Hopefully it makes its way over to Brew soon.

https://github.com/Homebrew/homebrew-cask/pull/56802

chrisabruce, 2019-02-14:
TL;DR workaround

```bash
brew tap PX4/homebrew-px4
brew update

#list available
brew search px4

#latest as of this post
brew install gcc-arm-none-eabi-80
```

therealprof, 2019-02-14:
Any reason not to follow the instructions at https://rust-embedded.github.io/book/intro/install/macos.html ?

therealprof, 2019-02-14:
For me it would make sense to amend the book with missing information (if any) and then just point to the book.

ests, 2019-03-09:
I guess it's also possible to just follow instructions for Linux "other distros" ([link](https://docs.rust-embedded.org/discovery/03-setup/linux.html#other-distros)), but of course just download package for Mac OS X 64-bit.

kylestew, 2019-03-21:
Seems like there are enough solutions here to make this a non-issue.

closed 2019-03-21 by kylestew

lead-free, 2020-10-21:
Just tried `brew install gcc-arm-embedded` and besides installing the toolchain brew started pulling Keil.. ctrl+c, went to ARM website to get it:(