rust-embedded/discovery #172: Sementation fault: 11

chrisabruce, 2019-02-14:
Not sure why I am getting this only on `--release`.  Non release version has no issue.

OSX 10.14
toolchain:  (brew) px4/px4/gcc-arm-none-eabi-80
rust ver: rustc 1.32.0 (9fda7c223 2019-01-16)

```bash
Chriss-Mac-Pro:06-hello-world chris$ cd ../07-registers/
Chriss-Mac-Pro:07-registers chris$ cargo run --release
   Compiling panic-itm v0.4.0
   Compiling aux7 v0.1.0 (/Users/chris/dev/discovery/src/07-registers/auxiliary)
   Compiling registers v0.1.0 (/Users/chris/dev/discovery/src/07-registers)
    Finished release [optimized + debuginfo] target(s) in 1.91s
     Running `arm-none-eabi-gdb -q -x openocd.gdb /Users/chris/dev/discovery/target/thumbv7em-none-eabihf/release/registers`
Reading symbols from /Users/chris/dev/discovery/target/thumbv7em-none-eabihf/release/registers...
Segmentation fault: 11
```

mathk, 2019-03-06:
looks like the issue is from gdb. How did you install gdb?

tpwrules, 2019-06-25:
I have this issue too. I installed it through `brew install armmbed/formulae/arm-none-eabi-gcc` which installs the toolchain from [here](https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-mac.tar.bz2).

dseverov, 2019-11-10:
> I have this issue too. I installed it through `brew install armmbed/formulae/arm-none-eabi-gcc` 
I have this issue too. Exactly.

dseverov, 2019-11-10:
It seems solution is there
https://github.com/ARMmbed/homebrew-formulae/pull/27