rust-embedded/discovery #178: 07 Registers --release Status_Access_Violation

michaeljprantl, 2019-03-09:
Hi,

I am reading through the material with great interest. I am currently in chapter 7.2 in the discovery book. I face the following problem when trying to build the --release:
`cargo run --release
    Finished release [optimized + debuginfo] target(s) in 0.24s
     Running `arm-none-eabi-gdb -q -x openocd.gdb [...]\target\thumbv7em-none-eabihf\release\registers`
[...]\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from [...]\target\thumbv7em-none-eabihf\release\registers...
error: process didn't exit successfully: `arm-none-eabi-gdb -q -x openocd.gdb [...]\target\thumbv7em-none-eabihf\release\registers` (exit code: 0xc0000005, STATUS_ACCESS_VIOLATION)`

Windows 10 Build 1803
toolchain: arm-none-eabi-gdb
Rust version: rustc 1.33.0 (2aa4c46cf 2019-02-28)

thejpster, 2019-06-19:
I also see this with on a thumbv6-none-eabi project:

```
$ rustc --version
rustc 1.35.0 (3c235d560 2019-05-20)
$ arm-none-eabi-gdb --version
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

thejpster, 2019-06-19:
I should also note that `gdb-multiarch` running under WSL works fine. arm-none-eabi-gdb 8-2018-q4-major for Linux (under WSL) gives a segfault. So it seems like a bug in that release of GDB?

tetious, 2019-06-29:
I have the same problem with 8-2018-q4-major. 

GNU Arm Embedded Toolchain: 7-2018-q2-update seems to work fine for me on Windows. This bundle seems to ship GDB 8.1, which is mentioned as the newest tested version in the book:

```GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 8.1.0.20180315-git```

PvdBerg1998, 2019-07-11:
Same issue on nightly win10, only when compiling with `--release`. Debug builds work.

PvdBerg1998, 2019-07-11:
It works for me if `lto = false` is set in `Cargo.toml`.

Okarin99, 2019-09-06:
It works with the newest version of the GNU-Toolchain. So I think the issue can be closed.
```
$ arm-none-eabi-gdb --version
GNU gdb (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

therealprof, 2019-09-06:
Yeah, the GNU toolchains from ARM are a bit hit and miss. One release works fine, the next release has a few severly broken tools which are fixed in the next one but then other stuff is broken. Not sure we can even recommend a specific release which works fine because it really depends on which jobs you're doing.