rust-embedded/discovery #184: No  led-roulette.exe file found

TonyScarpelli, 2019-05-11:
I downloaded the the rust-embedded/discovery code in the discovery repository as a Discovery-Master.Zip file.
I extracted it into a directory on my C-drive: "C:\Users\ascar\Downloads\Rust\Discovery Board"
I copied the "discovery-master" directory into "C:\Discovery"

I opened my Command Prompt, went to "C:\Discovery\src\05-led-roulette>" directory.
I entered "rustup target add thumbv7em-none-eabihf"
It came back with 
"info: component 'rust-std' for target 'thumbv7em-none-eabihf' is up to date"I went to
"C:\Discovery\src\05-led-roulette" directory and entered
"cargo build --target thumbv7em-none-eabihf"
After Downloading various files, and Compiling them it finished with
Compiling led-roulette v0.1.0 (C:\Discovery\src\05-led-roulette)
   Finished dev [unoptimized + debuginfo] target(s) in 1m 03s

Entered "cargo readobj --target thumbv7em-none-eabihf --bin led-roulette -- -file-headers"
Came back with
    Finished dev [unoptimized + debuginfo] target(s) in 0.13s
Error reading file: led-roulette.exe: no such file or directory

Got a directory: "C:\Discovery\target\thumbv7em-none-eabihf", though.
So what happened to the led-roulette.exe file?
Thanks.

therealprof, 2019-05-11:
.exe sounds wrong... Could be a problem with `cargo-binutils` blindly taking the executable ending from the OS.

TonyScarpelli, 2019-05-12:
**Compiling led-roulette v0.1.0 (C:\Discovery\src\05-led-roulette)**
**Finished dev [unoptimized + debuginfo] target(s)**

**Error reading file: led-roulette.exe: no such file or directory**

I'm following the book and just entering the code from the book (**Discover the world of microcontrollers through Rust!**)
I'm new at this, so I'm not sure what to do.
Are the book instructions too old for this version of rust that I am using? **cargo 1.34.0 (6789d8a0a 2019-04-01)**

<<.exe sounds wrong... Could be a problem with cargo-binutils blindly taking the executable ending from the OS.>>
What do you mean ".exe sounds wrong"? Should it be another file type?
And what do I do with "cargo-binutils"?  Should I re-install rust or something else?
Thanks.

closed 2019-05-12 by TonyScarpelli

therealprof, 2019-05-12:
> What do you mean ".exe sounds wrong"? Should it be another file type?

Yes, the filetype usually is ELF and it doesn't have a file suffix. It seems as if the tools automatically attache the `.exe` to the name because it is run on Windows. Linux and macOS don't use legacy file endings to determine the filetype.

arfa123, 2019-07-20:
I am also facing the same problem on:
Windows 10,
cargo 1.36.0, rustup 1.18.3, rustc 1.36.0
What is the solution?

arfa123, 2019-07-20:
install latest version of cargo-binutils instead of v 0.1.4. the problem will solved.

GregWoods, 2019-12-09:
The problem is still present in v 0.1.7

kevswims, 2019-12-20:
I am also seeing this problem in v 0.1.7. Is there a workaround for this?

Niondir, 2020-03-08:
Same here

```
$ cargo readobj --version
cargo-readobj 0.1.7
```

tomioe, 2020-04-14:
As said by arfa123, the correct command on Windows should be:

`cargo readobj target\thumbv7em-none-eabihf\debug\led-roulette -- -file-headers`

Also, since Windows doesn't handle symlinks the same way *NIX does, you need to remove the "target" file and run this command with admin priviledges (in the `src\05-led-roulette` folder):

mklink /D target c:\dev\rust_embedded\discovery\target

(On Windows, the command above should be run in every directory containing a "target" file, after removing it.)

tomasz-rozanski, 2020-04-14:
> As said by arfa123, the correct command on Windows should be:
> `cargo readobj target\thumbv7em-none-eabihf\debug\led-roulette -- -file-headers`

If you upgrade to `cargo-binutils v0.2.0`, you can use the original command again:
```
cargo readobj --target thumbv7em-none-eabihf --bin led-roulette -- -file-headers
```

> Also, since Windows doesn't handle symlinks the same way *NIX does, you need to remove the "target" file and run this command with admin priviledges (in the `src\05-led-roulette` folder):
>
> `mklink [...]

That's not entirely true. On Windows 10 `v1909` or newer (I believe), with the `Developer Mode` turned ON, symbolic links should work right off the bat.

tomioe, 2020-04-14:
> That's not entirely true. On Windows 10 `v1909` or newer (I believe), with the `Developer Mode` turned ON, symbolic links should work right off the bat.

Thanks, I was not aware of this. Do you think this should be added to the guide itself?

> If you upgrade to cargo-binutils v0.2.0, you can use the original command again:

Didn't have a chance to try this out, will definitely check it out!

Also, I still put this information in, because I think a lot of people experience this error when they run this the first time (and just follow the instructions "blindly").

tomasz-rozanski, 2020-04-14:
> 
> Thanks, I was not aware of this. Do you think this should be added to the guide itself?
> 
Maybe a link to this [article](https://github.com/git-for-windows/git/wiki/Symbolic-Links), from the official `git-for-windows` wiki, should be included in chapter 3.2: `Setting up a development environment -> Windows`. I'm open to suggestions.

Also, I'm not a 100% sure, if turning ON the `Developer Mode` alone does the trick, since I did my setup a few months ago. Feel free to verify my assumptions.

P.S. Turns out, this feature was introduced in Windows 10 `v1703`.

tomioe, 2020-04-15:
> Maybe a link to this [article](https://github.com/git-for-windows/git/wiki/Symbolic-Links), from the official `git-for-windows` wiki, should be included in chapter 3.2: `Setting up a development environment -> Windows`. I'm open to suggestions.
> 
> ~Also, I'm not a 100% sure, if turning ON the `Developer Mode` alone does the trick, since I did my setup a few months ago. Feel free to verify my assumptions.~ (I checked; it works).
> 
> P.S. Turns out, this feature was introduced in Windows 10 `v1703`.

I switched my Developer Mode to `On`, but [the symlinks that are included in the repo](https://github.com/rust-embedded/discovery/blob/master/src/05-led-roulette/target) are still not working. I am however now able to create symlinks without needing a privileged command prompt. Running on Windows 10 `v1903`.

See attached image for details.
[image]

tomasz-rozanski, 2020-04-15:
> I switched my Developer Mode to `On`, but [the symlinks that are included in the repo](https://github.com/rust-embedded/discovery/blob/master/src/05-led-roulette/target) are still not working. I am [...]

You're right, `cd target` command is not working in Windows without an extra step - my bad. The good news is, `cargo` and `cargo-binutils` are able to properly resolve the full path, using the original symlink, so you can use commands like this:
```
cargo run --bin led-roulette --target thumbv7em-none-eabihf
```

BTW: Windows `v1903` means, you did the last update over a year ago. Just saying 😎