rust-embedded/discovery #193: Find register addresses: why offset 0x18 ?

JPFrancoia, 2019-09-22:
Hi,

I'm kinda new to electronics and Rust, so please take that into account when answering.
I'm reading the Discovery book (great material by the way), and I'm at chapter 7. I would like to understand how to read the documentation, and how to find proper information about the Discovery card.

I understood that the LEDs used for the LED roulette are the LEDs LD3 to LD10. I understood that all these leds are connected to pins in the E block. I understood that the `GPIOE` peripheral's register block starts at `0x4800_1000`. I fund the documentation of the peripheral in the reference manual:

```
Section 11.4.12 GPIO register map - Page 243
```

But there is something I don't understand:

> We are interested in the register that's at an offset of 0x18 from the base address of the GPIOE peripheral. According to the table, that would be the register BSRR

I don't understand where the offset value comes from. Why is it `0x18`, specifically? This offset helps finding out that the register we're interested in is `BSRR`.

Could you please guid me through the reasoning?

adamgreig, 2019-09-22:
The code wants to write to `BSRR`, which is a register to set and reset pins on the GPIO port. To write to the register, the code has to know the register's memory address. We can look up `BSRR` in the reference manual:

[image]

and we see it says the offset is `0x18`, so we add that to the peripheral base `0x4800_1000` to get the register address `0x4800_1018`.

To be clear: the desire to use `BSRR` came first, _then_ we look up `BSRR` in the reference manual to find its offset. It's presented as a magical address in the book to show that registers are just integers at a certain memory address, but if you were writing the code from scratch you'd look up the appropriate register first and then find its address. Does that help explain?

JPFrancoia, 2019-09-23:
Aaaah, yes, it definitely helps. Sorry I completely missed that. The starter code in the chapter clearly mentions `const GPIOE_BSRR: u32 = 0x48001018`.

Ok, I'll now try to ask a less naive question: how would I find it's the `BSRR` register in the first place?

GPIO Registers -> GPIO port bit set/reset register (GPIOx_BSRR) ?

adamgreig, 2019-09-23:
> how would I find it's the BSRR register in the first place?

I'm afraid I don't really understand the question. If you mean "how do I find what register to use to set pins", then the answer is look through the reference manual, read the description of the peripheral which often mentions how to use it, and then read through the description there of all the registers.

If you mean "how do I find that 0x4800_1018 is BSRR", then the register map in the reference manual is probably more useful: the memory map near the start will say 0x4800_1000 is GPIOE, and the GPIO register map will say 0x18 is BSRR.

JPFrancoia, 2019-09-23:
> I'm afraid I don't really understand the question. If you mean "how do I find what register to use to set pins", then the answer is look through the reference manual, read the description of the [...]

That's what I meant, yes. It's just not super easy to go through all the process:

- Finding out that the LEDs are connected to the GPIO E
- Finding out how to set the pins of the GPIO E
- Section 11.4.7 explains how to set some registers that in turn will set GPIO pins, but wait it's the same for ports A, B, C, D, E
- Let's look in the registry map, in section 11.4.12
- I see there I need to look in section 3.2.2 for the boundary addresses
- finding the boundary addresses for the GPIO E
- adding the offset (to access BSRR for the GPIO E ?)

(please correct me if it's not the right way to find the information: the right register to modify).

I think I kind of got it, but I think the book is slightly misleading on this point, since it directly gives the magic address (but I suspect it was for simplicity's sake).

closed 2019-09-25 by JPFrancoia

NitinSaxenait, 2021-06-21:
After blinking the LEDs. I started with the chapter register and on the first page itself I got some issues. I read about most of them and find out that the address is const and to use that fixed value we used raw pointer `GPIOE_BSRR as *mut u32` which is making a copy of the address(referencing) variable and allowing us to change.
Second, we are dereferencing using * and putting the statement inside the unsafe block as per the raw pointer deference rules..

I hope till here i am getting it right now the next issue is what we are doing at right-hand side using the shift operator. We are using 9 11 to set and 16, 25 to reset **but why 1 in every shift?**

`#![no_main]
#![no_std]

#[allow(unused_imports)]
use aux7::{entry, iprint, iprintln};

#[entry]
 fn main() -> ! {
    aux7::init();

   {
        // A magic address!
        const GPIOE_BSRR: u32 = 0x48001018;

        // Turn on the "North" LED (red)
        //we use the dereference operator * on a raw pointer that requires an unsafe block.
        //change data through the mutable pointer
        (GPIOE_BSRR as *mut u32) = 1 << 9;

        // Turn on the "East" LED (green)
        *(GPIOE_BSRR as *mut u32) = 1 << 11;

        // Turn off the "North" LED
        (GPIOE_BSRR as *mut u32) = 1 << (9 + 16);

        // Turn off the "East" LED
        (GPIOE_BSRR as *mut u32) = 1 << (11 + 16);
    }
    loop {}
}
`