rust-embedded/discovery #194: Finding the pins for I2C in chapter 15

JPFrancoia, 2019-09-28:
Hi,

I'm new to Rust and electronics and I struggle to understand a couple of things.

In the `aux` library of chapter 15 of the discovery book, the init function initializes the I2C sensor.
I can't figure out how to find out why the GPIO B, pins 6 and 7, were chosen for the communication.

Here is the block that gives me trouble:

```rust
    let mut gpiob = dp.GPIOB.split(&mut rcc.ahb);
    let scl = gpiob.pb6.into_af4(&mut gpiob.moder, &mut gpiob.afrl);
    let sda = gpiob.pb7.into_af4(&mut gpiob.moder, &mut gpiob.afrl);
```

Can someone explain to me why we use the GPIO B, pins 6 and 7 ? Could we have chosen any port/pin?

I also don't understand the `into_af4` function. The doc just says:

> Configures the pin to serve as alternate function 4 (AF4)

Why did we chose AF4 here? (why not AF5 or AF6 for example?).

Cheers

therealprof, 2019-09-30:
Hi @JPFrancoia,

> Can someone explain to me why we use the GPIO B, pins 6 and 7 ? Could we have chosen any port/pin?

These STM32 chips only support certain functions on certain pins. The example probably could have used alternative pins (potentially also for a different I2C peripheral, there typically more than just one on these chips).

> I also don't understand the into_af4 function. The doc just says:

What AF4 means depends on the specific pin in question. You can checkout the meaning and options for the STM32F303 used on this board in the datasheet here (p. 45ff): https://www.st.com/resource/en/datasheet/stm32f303vc.pdf

JPFrancoia, 2019-09-30:
aaaah ok, I need to look into "pinouts and pin description"...

Thanks

closed 2019-09-30 by JPFrancoia