rust-embedded/discovery #199: RTRM question

markem1, 2019-11-27:
On this page, https://docs.rust-embedded.org/discovery/07-registers/rtrm.html is this statement:

We are interested in the register that's at an offset of 0x18 from the base address of the GPIOE peripheral. According to the table, that would be the register BSRR.

Why are we interested in this offset?  How does one conclude that this is the correct offset / register for our uses.  I've RTRM's a lot and cannot find an obvious connection.

markhildreth, 2020-04-28:
@jamesmunns 

I'm planning on working on some PRs that will hopefully alleviate these types of question, as I agree with one redditor who said that is seems like the book "skipped a chapter." I have come up with the following high-level plan for the curious to allow for early feedback for those curious.

1.) Add a section in "Meet Your Hardware" to better explain how the microcontroller and board interact.

Reasoning: This will hopefully make it more apparent that we are a.) programming the microcontroller, which itself is b.) changing how the pins are operating. These pins are connected to the board in such a way that they can c.) drive the LEDs on the board. This context should be helpful for Chapter 7.

2.) Simplify the introduction to chapter 7, restricting it to just the new code, a quick description of what the code does, and a promise that we'll explain how it works over the course of the chapter.

Reasoning: Currently, the intro tries to explain some of the code in a few paragraphs, with some quick text on registers/addresses. This makes it seem like that is all the tutorial is going to talk about how the code works. The "But how can one find out what this register does?" line is in a spot that is easy to miss between the end of the pin-driving aside and the end of the page, which might add to the confusion.

3.) In chapter 7, go a _bit_ more into detail on memory mapping and registers. This will explain what a register is, why we need to access it through memory address, and generally set up the idea that in order to do anything with the LEDs we need to manipulate registers in a specific way.

Reasoning: I feel it's important to ensure developers understand that even though `0x12345678` and `0x87654321` look like they both access "memory", unlike typical programming, we can't assume that all of this memory works and acts the same. Section 7.3 (`0xBAAD address`) talks about this a bit, but I feel we've already lost some people by then.

4.) Much more explanation on how the author was able to determine all the info needed to drive the LED through the BSRR register in the "RTRM" section.

Reasoning: Currently, we write that we found specific information on a specific page of a reference manual, but little about WHY we looked at that page or for that information. That is the knowledge that is needed in order to complete the next chapter's challenge, so it's important to explain it.

jamesmunns, 2020-05-03:
@markhildreth I totally agree with your plan, and wanted to thank you again for writing these up! Let me know if there is anything I can do to help, and I'll try to keep all of your PRs reviewed as quickly as possible!