rust-embedded/discovery #200: Cannot connect to HC-05 serial port on Win 10

mjptree, 2020-01-01:
Hi team,

The information on how to connect Putty to the HC-05 is a bit sparse. There are several sources of potential issues, where it is **really** difficult to research any help:

1. When coupling the Bluetooth device on your machine with the HC-05, Windows does not appear to automatically assign a COM port to the connection. 
2. Apparently it is possible to manually assign COM ports to the connection, but there is little information, whether the required one is the outgoing or the incoming connection. Either way, the `mode` command, does not show the manually assigned COM ports.
3. There is no information about the proper configuration of the port, whether one should reuse the configuration of the serial-to-USB converter - other than the below code comment:

https://github.com/rust-embedded/discovery/blob/98a7bd44fd79eeb8e7e8bd29613b5d0b9b76dc4f/src/11-usart/auxiliary/src/lib.rs#L31

4. Putty only successfully connects to the incoming COM port, but when finally managing to open the serial monitor - and of course having wired up the loopback configuration - the monitor does not echo back any typed input.

Can you please advise?

Thanks and happy new year!
Michael

bahrozjaved, 2020-04-30:
you had to check all the configuration there is some mistake that could be happpened during the setup of hc-05

pevandenburie, 2021-03-15:
I ran into this issue. Here is my solution (real wording might differ as I run Windows in FR) :

1. Open Window Bluetooth parameters, and click on "Add a new BT device"
2. You have 3 choices: "BT device", "Screen wireless dock" and "Other". Choose the first ("BT device"). After few seconds **DSD TECH HC-05** should appear: pair-it using default code 1234. 
3. COM port is not visible when typing 'mode' command. Still from Window Bluetooth parameters screen, go to "Advanced BT parameters". In COM Ports section, you should see created COM ports:
    - COM4  -  Input  - DSD TECH HC-05
    - COM5  -  Output - DSD TECH HC-05 'Dev B'
4. Using Putty, open a serial line on COM4 (the Input port) at 9600 bauds (the default one when you didn't change anything).

pfesenmeier, 2021-07-24:
Thanks @pevandenburie! Windows assigned COM5 as the output and COM4 as input on my machine. Regardless, connecting to the output port worked for me.