rust-embedded/discovery #203: arm-none-eabi-gdb error

carriere4, 2020-01-31:
Getting this error message when I run `arm-none-eabi-gdb` to test the debugger for the first time:
```
“arm-none-eabi-gdb” cannot be opened because the developer cannot be verified.
macOS cannot verify that this app is free from malware.
```

jacobrosenthal, 2020-01-31:
Mac is becoming increasingly developer hostile. Any googling will show you
common possible fixes are using sudo, disabling SIP protection, and or
codesigning gdb.

carriere4, 2020-01-31:
Wow! Thanks so much for the prompt reply. Will try those and if anything works, I'll report back. I've got a Linux machine, so I'll try that as well.

carriere4, 2020-02-09:
works just fine on Linux, so will continue on that path. Will come back to this issue + open and update if I get a fix on the Mac side of things.

closed 2020-02-09 by carriere4