rust-embedded/discovery #206: itmdump not functioning as shown in 06-hello-world

ARMATAV, 2020-02-08:
NOTE: **This is running both OpenOCD and itmdump in the `/tmp` directory on macOS Catalina 10.15.3**

1.  touch `itm.txt`
2. `openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg`
3. `itmdump -F -f itm.txt`
4. Navigate to `06-hello-world` in a separate terminal window
5. `cargo run` after modifying `config` to set the target
6. `step` into `main()` and `next` until `loop {}` is highlighted
7. Nothing is sent over itmdump to the text file, `itm.txt` is empty

Here's what the terminal looks like after the fact

**OpenOCD** (`/tmp`)

```
❯ openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v15 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.884235
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
```

**gdb** (`.../src/06-hello-world`)

```
Breakpoint 1, main () at src/06-hello-world/src/main.rs:8
8       #[entry]
(gdb) monitor tpiu config internal itm.txt uart off 8000000
(gdb) monitor itm port 0 on
(gdb) step
hello_world::__cortex_m_rt_main () at src/06-hello-world/src/main.rs:10
10          let mut itm = aux6::init();
(gdb) next
12          iprintln!(&mut itm.stim[0], "Hello, world!");
(gdb) next
14          loop {}
(gdb) 
```

**itmdump** (`/tmp`)

```
❯ itmdump -F -f itm.txt
❯
```

Expected behavior;

itmdump shows `Hello, world!` or some other processed input after we reach `iprintln!`

Instead, it's blank and `itm.txt` is empty;

```
❯ cat itm.txt 
❯ 
```

It seems like itmdump is not functioning properly with the setup described in `06-hello-world`, but I must be missing something, what am I missing?

ARMATAV, 2020-02-16:
I'm totally trolling and literally scrolled past the big image that explains how to fix this. Flip the board over if you find this issue because you too cannot read basic instructions like me;

If `SB10` looks like **<** instead of ⬛- then you're screwed and need to jump `SB10` the second image below.

[image]

[image]

closed 2020-02-16 by ARMATAV

mlh758, 2020-04-12:
I'm at the same step of the tutorial and my SB10 is not soldered. I figured I would go with the jumper since that's easier but going from that last pin on SWD (mine says SWD instead of SWO in the book??) to PB3 leaves me the with the same results. Are there any further steps for troubleshooting around anywhere or should I just try soldering SB10?

TASSDevon, 2021-06-29:
Not sure why but I connected PB3 with SWO and I'm not getting anything with itmdump. Any ideas what could still be wrong? Already tried different jumper cables to rule those out. Bought the board a few weeks ago, SB10 is not soldered.

maxdignan, 2021-07-17:
Ran into this issue as well. Thank you for making this post @ARMATAV !

zanayr, 2021-09-23:
Hello,
Running into a similar issue. I don't have the SB10 soldered and I have the dupont cable running from the last pin in the SWD line to PB3 pin.

I started the OpenOCD in the chapter 6 directory, then ran the two itmdump commands.
```
cd /tmp && touch itm.txt
itmdump -F -f itm.txt
```

This creates the text file in the 06-hello-world directory and the program is writing to the file, but I'm not seeing anything in the itm terminal. Any ideas?

maxdignan, 2021-09-23:
Hey @zanayr, I am not sure that the syntax calling itmdump is correct. I've always done it with only one of the F's, but I could be wrong on that. I ended up having better luck just reading from the file in a text viewer/editor like vscode. I'd suggest trying that before double-checking and rewiring what you have so far.

analyst1001, 2021-11-16:
I ran into a similar issue while following the instructions for `06-hello-world`. I am using a Windows 10 setup, and I connected SWO to PB3 using female to female jumper wire. I observed an error message when I connected GDB client to GDB server:

OpenOCD output before connecting GDB client:
```
> openocd -s C:\share\scripts -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.   
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : Listening on port 6666 for tcl connections   
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J37M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 2.906982
Info : stm32f3x.cpu: Cortex-M4 r0p1 processor detected      
Info : stm32f3x.cpu: target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 3333
Info : Listening on port 3333 for gdb connections  
```

OpenOCD output when I connected GDB client to the server:
```
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x10036422
Info : flash size = 256kbytes
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000194 msp: 0x2000a000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000194 msp: 0x2000a000
Info : DEPRECATED 'tpiu config' command: Adding a TPIU 'stm32f3x.cpu.tpiu' in the configuration
Info : DEPRECATED 'tpiu config' command: Running: 'tpiu create stm32f3x.cpu.tpiu -dap stm32f3x.dap -ap-num 0'
Info : DEPRECATED 'tpiu config' command: Running: 'stm32f3x.cpu.tpiu configure -protocol uart -traceclk 8000000 -output itm.txt -formatter off'
Info : DEPRECATED 'tpiu config' command: Running: 'stm32f3x.cpu.tpiu enable'
Error: SWO pin frequency not set

Info : halted: PC: 0x080001f6
```

I noticed the message `Error: SWO pin frequency not set` in here, and decided to set the pin frequency manually to `2000000` (I used this value arbitrarily because this was the max supported value for me) using the command `monitor tpiu config internal itm.txt uart off 8000000 2000000` in GDB client:

```
Breakpoint 1, hello_world::__cortex_m_rt_main_trampoline () at src\06-hello-world\src\main.rs:8
8       #[entry]
halted: PC: 0x080001f6
hello_world::__cortex_m_rt_main () at src\06-hello-world\src\main.rs:10
10          let mut itm = aux6::init();
(gdb) monitor tpiu config internal itm.txt uart off 8000000 2000000
DEPRECATED 'tpiu config' command: Using stm32f3x.cpu.tpiu
DEPRECATED 'tpiu config' command: Running: 'stm32f3x.cpu.tpiu configure -protocol uart -traceclk 8000000 -pin-freq 2000000 -output itm.txt -formatter off'
DEPRECATED 'tpiu config' command: Running: 'stm32f3x.cpu.tpiu enable'
(gdb)
```

OpenOCD output after manually entering the monitor command:

```
Info : halted: PC: 0x080001f6
Info : DEPRECATED 'tpiu config' command: Using stm32f3x.cpu.tpiu
Info : DEPRECATED 'tpiu config' command: Running: 'stm32f3x.cpu.tpiu configure -protocol uart -traceclk 8000000 -pin-freq 2000000 -output itm.txt -formatter off'
Info : DEPRECATED 'tpiu config' command: Running: 'stm32f3x.cpu.tpiu enable'
```

After this I continued with `next` command in GDB client and was able to observe the output from itmdump command.

```
(gdb) n
halted: PC: 0x0800024e
halted: PC: 0x08000202
halted: PC: 0x08000204
halted: PC: 0x08000206
12          iprintln!(&mut itm.stim[0], "Hello, world!");
(gdb) n
halted: PC: 0x08000208
halted: PC: 0x0800020c
halted: PC: 0x08000210
halted: PC: 0x08000214
halted: PC: 0x08000216
halted: PC: 0x0800021e
halted: PC: 0x0800021c
14          loop {}
(gdb)
```

```
> itmdump -F -f itm.txt
Hello, world!
```

To be honest, I am not sure why the existing monitor command did not work, since based on the description mentioned at https://openocd.org/doc/html/Architecture-and-Core-Commands.html, `tpiu configure` should select the maximum supported rate automatically for pin frequency.

wldmr, 2021-12-24:
Found this issue after googling `"swo pin frequency not set"`. My problem was the same as @analyst1001's (no output, using the jumper cable method). Their answer solved it.

I've attached a [patch file](https://github.com/rust-embedded/discovery/files/7774971/openocd.gdb.pin-frequency.patch.txt). It's simple enough, but I don't know enough about the subject matter to know whether that's an actual improvement, or just a workaround for our particular environments/boards.

If it's a good solution, I think the change is small enough that we don't need a pull request. However, if you prefer a PR I'll happily oblige.

```patch
diff --git a/f3discovery/src/openocd.gdb b/f3discovery/src/openocd.gdb
index 7dc8dd7..fc1d518 100644
--- a/f3discovery/src/openocd.gdb
+++ b/f3discovery/src/openocd.gdb
@@ -15,7 +15,7 @@ set style sources off
 
 # Initialize monitoring so iprintln! macro output
 # is sent from the itm port to itm.txt
-monitor tpiu config internal itm.txt uart off 8000000
+monitor tpiu config internal itm.txt uart off 8000000 2000000
 
 # Turn on the itm port
 monitor itm port 0 on
```

dobromyslov, 2022-01-11:
I confirm @analyst1001 comment helped me to solve the issue with `Error: SWO pin frequency not set` (SB10 soldered manually).

@wldmr Also need to fix the Book here:
https://github.com/rust-embedded/discovery/blob/8b6953b887617039a1dc25aa9a939eba27f1b8a8/f3discovery/src/05-led-roulette/the-challenge.md?plain=1#L52

It would be nice if you add a PR with your fixes for `openocd.gdb` files and for the Book also.