rust-embedded/discovery #215: Unable to build LED roulette first example

plusangel, 2020-02-27:
Dear all, 

I need your valuable on the following. I have tried it in both Win10 and Ubuntu18, but let me focus on Win10.

My versions are: 
```
rustc 1.41.0 (5e1a79984 2020-01-27)
itm v0.3.1
cargo-binutils v0.1.7
```

After executing:
`cargo build --target thumbv7em-none-eabihf`
I got the following output:
```
   Compiling semver-parser v0.7.0
   Compiling typenum v1.11.2
   Compiling proc-macro2 v1.0.8
   Compiling unicode-xid v0.2.0
   Compiling cortex-m v0.6.2
   Compiling stable_deref_trait v1.1.1
   Compiling syn v1.0.15
   Compiling vcell v0.1.2
   Compiling cortex-m-rt v0.6.12
   Compiling void v1.0.2
   Compiling nb v0.1.2
   Compiling cortex-m v0.5.10
   Compiling stm32f30x v0.7.1
   Compiling aligned v0.2.0
   Compiling r0 v0.2.2
   Compiling f3 v0.6.1
   Compiling panic-halt v0.2.0
   Compiling volatile-register v0.2.0
   Compiling embedded-hal v0.2.3
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling quote v1.0.2
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling generic-array v0.11.1
   Compiling l3gd20 v0.2.0
   Compiling as-slice v0.1.3
   Compiling lsm303dlhc v0.2.0
   Compiling aligned v0.3.2
   Compiling cortex-m-rt-macros v0.1.8
   Compiling stm32f30x-hal v0.2.0
   Compiling aux5 v0.1.0 (C:\Users\angelos\Documents\discovery\src\05-led-roulette\auxiliary)
   Compiling led-roulette v0.1.0 (C:\Users\angelos\Documents\discovery\src\05-led-roulette)
    Finished dev [unoptimized + debuginfo] target(s) in 58.05s
```

Then I move to the target folder where I can find it in the root of the repo, similar to that:
`C:\Users\angelos\Documents\discovery\target\thumbv7em-none-eabihf\debug`

and then when I run:
` cargo readobj --target thumbv7em-none-eabihf --bin led-roulette -- -file-headers`
I got 
```
   Compiling stable_deref_trait v1.1.1
   Compiling vcell v0.1.2
   Compiling void v1.0.2
   Compiling nb v0.1.2
   Compiling aligned v0.2.0
   Compiling r0 v0.2.2
   Compiling panic-halt v0.2.0
   Compiling typenum v1.11.2
   Compiling volatile-register v0.2.0
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling embedded-hal v0.2.3
   Compiling cortex-m-rt v0.6.12
   Compiling generic-array v0.12.3
   Compiling generic-array v0.13.2
   Compiling generic-array v0.11.1
   Compiling l3gd20 v0.2.0
   Compiling lsm303dlhc v0.2.0
   Compiling as-slice v0.1.3
   Compiling aligned v0.3.2
   Compiling cortex-m v0.6.2
   Compiling cortex-m v0.5.10
   Compiling stm32f30x v0.7.1
   Compiling stm32f30x-hal v0.2.0
   Compiling f3 v0.6.1
   Compiling aux5 v0.1.0 (C:\Users\angelos\Documents\discovery\src\05-led-roulette\auxiliary)
   Compiling led-roulette v0.1.0 (C:\Users\angelos\Documents\discovery\src\05-led-roulette)
    Finished dev [unoptimized + debuginfo] target(s) in 30.00s
error: missing field `package`
```

Then obviously, even I tried to flash it's not working...

Any ideas? Do you need more details? What I'm doing wrong?
I have tried various boards like F3 but also STM32F103 - and get exactly the same behaviour.

Rosto75, 2020-02-27:
This command is looking for a `package` section from the `cargo.toml` file. Just run it from the project folder, `src\05-led-roulette`.

plusangel, 2020-02-28:
I tried both and I get:
```
C:\Users\angelos\Documents\discovery\src\05-led-roulette>cargo readobj --target thumbv7em-none-eabihf --bin led-roulette -- -file-headers
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
error: The directory name is invalid. (os error 267)
```

tomaszrozanski, 2020-03-02:
Do you have your code on GitHub? It would be a huge help to be able to analyze the whole project.

plusangel, 2020-03-07:
No, I just follow the tutorial. However, I was able to experiment using the cortex-m-quickstart and importing the led-roulette code

tomaszrozanski, 2020-03-07:
Clearly, there's a problem with symbolic links in your system. Check if the *Developer Mode* is turned on in Windows. 
Until then, try this workaround:
```sh
cargo readobj ..\..\target\thumbv7em-none-eabihf\debug\led-roulette -- -file-headers`
```