rust-embedded/discovery #216: Section 6 (Hello World) should have a section showing what soldered/unsoldered looks like.

TovarishFin, 2020-02-29:
As a complete beginner trying to do this tutorial, I spent quite a bit of time trying to get `iprintln!` to work. [issue 206](https://github.com/rust-embedded/discovery/issues/206) has an image attached which makes it clear what unsoldered and soldered looks like. After seeing that image, it was clear that I needed to use a jumper wire. It would be great to put that image into the the book in order to make things that much more clear.

David-OConnor, 2020-04-01:
Same. Mine was as in the image, had silver (solder?) on it (ie not like the exposed copper ones), but didn't work without the cable.

bahrozjaved, 2020-04-30:
you haade to attach the jummper wire form the itm dum print or if you wnat to run the hello world withour jumper then you need to do some soldering work by seeing the artitecture of the board