rust-embedded/discovery #219: Verification on Windows report

David-OConnor, 2020-03-30:
On the [Verification](https://docs.rust-embedded.org/discovery/03-setup/verify.html) step where you open the OpenOCD connection, I've noted the following:

The command labeled as for `*nix`: `openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg` works, although with a warning to use `stlink.cfg`, ie without the `-v2-1`, and the one labeled for windows doesn't work; it raises this error:
`$: The term '$' is not recognized as the name of a cmdlet, function, script file, or operable pr
ogram.`

adamgreig, 2020-03-31:
In this case the `$` is meant to indicate that this line is a command for you to execute; you shouldn't actually type the `$`. It's used because on UNIX/Linux/Mac shells a `$` symbol is the historical prompt symbol.

On windows as on *nix you should just run the command itself: `openocd -s C:\OpenOCD\share\scripts -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg`

David-OConnor, 2020-03-31:
Thank you very much. Either command now works for me.

closed 2020-03-31 by David-OConnor