rust-embedded/discovery #221: Change breakpoint cmd

David-OConnor, 2020-03-30:
[Page](https://docs.rust-embedded.org/discovery/05-led-roulette/debug-it.html)

Instructions say `break main` from gdb. This sets the break at line 7, the `#[entry]` line. I've found to get it to work (Windows), the command should be `break main.rs:10`

adamgreig, 2020-03-31:
Was there any problem with `break main`? The `#[entry]` line should still end up being the start of your main function, so debugging would hopefully work OK. It's generally possible to add breakpoints with either the name of a function or a specific line number; if possible it's better for the tutorial to use the function name in case the line numbering changes (e.g. due to an extra comment or import or whatever).

David-OConnor, 2020-03-31:
It works, if you use a second `continue`. (Presumably to jump into the function, then the main loop, while in the example, the `break main` puts you in the fn instead of just outside it?)

bahrozjaved, 2020-04-30:
in gdb the continue is the command that can run the program at the end of the fn where there is infinite loop so the program is stuck in infinite loop and runs infinite times