rust-embedded/discovery #222: Suggestion

David-OConnor, 2020-04-05:
I'm too new/bad at this to PR atm, so maybe later. I think a good bet for a follow-on lesson would be how to navigate the modules the `aux` modules depend on. It seems like there's a layer of general-specific Hal modules going on, and if you detangle, it gets pretty confusing due to how many layers there are, and that things are renamed. For example, the `f3` crate is a thin wrapper of `stm3230x_hal`, plus some high-level code. But the way it structures its imports is awkward. And for example, the variable `rcc` is used to mean different things/types depending on where you import it from. 

The `aux` module is great for the lessons since it abstracts things away, but we need a weaning off. For example, here's a detangle of some of the imports.

```rust
use stm32f30x_hal::{
    prelude::*,
    stm32f30x::{self, RCC, TIM6, rcc, tim6},
    gpio::gpioe
};
```

David-OConnor, 2020-04-05:
Here's what I've come up with, using an example from the `timers` module. Others should be similar. This is more general and current than `aux`:

Cargo.toml:
```toml
[dependencies]
cortex-m = "0.6.2"
cortex-m-rt = "0.6.12"
stm32f3xx-hal = { version = "0.4.2", features=["stm32f303xc", "rt"] }
panic-itm = "0.4.1"
```

Imports:
```rust
pub use cortex_m::asm::{bkpt, nop};
use cortex_m_rt::entry;

use stm32f3xx_hal as hal;
use hal::{
    prelude::*,  //stm32f30x imported in prelude?
    stm32,
    prelude::_embedded_hal_digital_OutputPin,
};

use panic_itm; // panic handler
```

bahrozjaved, 2020-04-30:
yeah you are right but what i can do with that project is that i can control the 5v relay module connected with stm32f3 and the bluetooth hc-05 also connected  through the jummper wirees and controlled my rooms lights that runs on 220v

David-OConnor, 2020-04-30:
After getting more experience with rust on embedded, and the `embedded-hal` ecosystem, I think a better approach would be to ditch the `aux` modules entirely, and use `hal` traits/modules directly, eg my above post . It'd make it more like a real program, and easier to dive from this book into a project. Abstractions are nice when they make things general, not specific.

bahrozjaved, 2020-04-30:
by seeing your repository I know that you are senior than me.Yes you are
right but I am new in rust programming language at the initial level and I
am. Leraning the embedded Hal crate that how it works so that's what this
is my first milestone of real life project so that why I can store it and
make public for all the developer who wanted to study rust.