rust-embedded/discovery #224: Trouble building 05 LedRoulette

ondono, 2020-04-14:
Runing:
`cargo build --target thumbv7em-none-eabihf`

is followed by a long list of messages repeating:
`'znver1' is not a recognized processor for this target (ignoring processor)`

I'm guessing that for some reason cargo is not correctly targeting the right processor. I've tried both in nightly and stable with the same result. Any clues?

adamgreig, 2020-04-14:
Do you have some environment variables set (check `env`) or a Cargo config that might be adding some extra rust or cargo flags, for example specifying -mcpu for compiling something for your (presumably AMD Zen) processor?

ondono, 2020-04-15:
Completely right. Turns out I was incorrectly sourcing a file from an old project that was setting RUSTFLAGS to "-C target-cpu=native". Thank you very much!

closed 2020-04-15 by ondono