rust-embedded/discovery #228: Serial Communication now working on MacOS

kianenigma, 2020-05-02:
- MacOS 10.15.4
- my board is new revision; I needed to connect BP3 is [explained here]( https://rust-embedded.github.io/discovery/06-hello-world/index.html) to enable `itdump`.

```
$ ls -l /dev/cu.*
crw-rw-rw-  1 root  wheel    9,   7 May  2 16:45 /dev/cu.usbmodem143403
```

and I actually have both `tty.x` and `cu.x`

```
ls /dev | rg "usbmodem143403"
cu.usbmodem143403
tty.usbmodem143403
```

- board connected via `USB-STLINK`.

- echo-ing into either of the above just hangs, and the terminal never exits anymore, even with ctrl-c. 

- During all stages, I have `openocd` open. 

- also, when `minicom` is open with the loopback enabled (shorting PC4 and PC5), I cannot see anything as I type. Consequently, I've tried section [`11.1`](https://rust-embedded.github.io/discovery/11-usart/send-a-single-byte.html) and the one byte is not transmitted back to the host 

Perhaps something with the macos setup is broken?

mlh758, 2020-05-09:
I'm also stuck at [section 10](https://docs.rust-embedded.org/discovery/10-serial-communication/nix-tooling.html). I think at this step we just need the USB to serial module plugged in, nothing to the discovery board. However, it's just hanging for me as well.

The echo/minicom commands freeze the terminal forever whether I'm using the cu or tty file under /dev.

Mine dev usb devices look like this:
```
❯ ls -l /dev | grep usb
crw-rw-rw-  1 root  wheel           18,   5 May  9 09:23 cu.usbserial-0001
crw-rw-rw-  1 root  wheel           18,   4 May  9 09:23 tty.usbserial-0001
```

wbuck, 2020-12-10:
How were you to connecting to the board? Through the ST-LINK port or the USB USER port?