rust-embedded/discovery #244: Can't connect to GDB Server

connorjcantrell, 2020-05-26:
I am having trouble connecting to the GDB Server using my MacOS. I have been following the instructions, but still having an issue. I am almost certain I should be using `arm-none-eabi-gdb`, but am receiving an error. I have tried the other alternative command, but with no success. Does anyone know what is wrong? 

_I am inside the `05-led-roulette` directory_

### arm-none-eabi-gdb
```
$ arm-none-eabi-gdb -q target/thumbv7em-none-eabihf/debug/led-roulette
/Users/myname/.gdbinit:2254: Error in sourced command file:
Scripting in the "Python" language is not supported in this copy of GDB.
target/thumbv7em-none-eabihf/debug/led-roulette: No such file or directory.
```

### gdb
```
$ gdb -q target/thumbv7em-none-eabihf/debug/led-roulette
target/thumbv7em-none-eabihf/debug/led-roulette: No such file or directory.
```

### gdb-multiarch
```
$ gdb-multiarch -q target/thumbv7em-none-eabihf/debug/led-roulette
-bash: gdb-multiarch: command not found
```

adamgreig, 2020-05-27:
The main error in all three cases seems to be that the file "target/thumbv7em-none-eabihf/debug/led-roulette" doesn't exist. Usually this would mean you haven't built it: perhaps you're building for the wrong target, or building with `--release` (so would use the `release` folder instead of `debug`), or there was an error building.

Could you try running `cargo build --target thumbv7em-none-eabihf` and check it succeeds, and then try checking all the files inside `target` to see if `target/thumbv7em-none-eabihf/debug/led-roulette` is created?

connorjcantrell, 2020-05-27:
> Could you try running cargo build --target thumbv7em-none-eabihf and check it succeeds, and then try checking all the files inside target to see if target/thumbv7em-none-eabihf/debug/led-roulette is created?

I just repeated the build process and checked for files inside `target`. **I found `target` to be type alias (12 bytes)** and not a folder. 

It appears that is my problem, but why did I get an alias filetype (symbolic link?) instead of a folder?

adamgreig, 2020-05-27:
It's normal for `target` to be a symlink; it's set up that way in this repository so that all the examples share a single target directory at the root of the repository. That shouldn't cause problems; the directory that target points to should contain your build objects.

connorjcantrell, 2020-05-27:
It appears to be a broken link
```
$ cd -L target
-bash: cd: target: No such file or directory
```

adamgreig, 2020-05-27:
That's also normal for a fresh checkout of the repository, but running `cargo build --target thumbv7em-none-eabihf` from the `05-led-roulette` directory should create that top-level target directory, so that `src/05-led-roulette/target` becomes a valid link.

Have you run the `cargo build --target thumbv7em-none-eabihf` command and did it complete successfully?

connorjcantrell, 2020-05-27:
Running `cargo build --target thumbv7em-none-eabihf` completed successfully

adamgreig, 2020-05-27:
Oh, sudden thought: did you check out the repository inside a cargo workspace, in other words with a Cargo.toml file in the parent directory to the repository checkout? If so, cargo would use a target directory in that parent repository, so the symlink would be wrong. I'd suggest just not having it in a workspace but otherwise you could adjust the symlink to be one level higher.

If it's not in a workspace, could you try to locate the build objects? Cargo won't have succeeded without writing them somewhere, so they ought to exist.

connorjcantrell, 2020-05-27:
I apologize if I am not understanding correctly. I changed to the parent directory (the cloned repository) and ran the following command: 
```
$ find . -name target
./src/07-registers/target
./src/06-hello-world/target
./src/16-punch-o-meter/target
./src/11-usart/target
./src/14-i2c/target
./src/05-led-roulette/target
./src/08-leds-again/target
./src/09-clocks-and-timers/target
./src/15-led-compass/target
```
This confirms that it does not exist in the parent directory. 

I'm sorry, but I don't know how to locate the build objects. How can I learn more about build objects?

adamgreig, 2020-05-27:
Sorry, I probably wasn't clear about the workspace thing: is there a Cargo.toml file or a target directory in the parent directory of the repository, in other words one level above the repository? Cargo supports a feature called workspaces where many crates work together and share a target; if you cloned this repository into a workspace it would cause it to use the workspace target directory, which would explain why you don't see a target directory inside the repository and why the symlinks are invalid. This repository isn't designed to be used inside a workspace.

If that's not the case, you might also have some leftover environment variables set from some other project which alter the behaviour of Cargo. If you run the `env` command it will list all environment variables; check any starting with RUST or CARGO that look suspicious.

connorjcantrell, 2020-05-27:
Ah! after running `env`, I found a `target` directory inside my home directory. Maybe we're getting somewhere?

`User/<username>/target` contains:
- `debug`
- `release`
- `thumbv7em-none-eabihf`

Why is my `symlink` invalid? Should I delete `User/<username>/target` and build a new target inside the parent directory?

adamgreig, 2020-05-27:
The symlink is broken because it always points to a target directory at the root of this repository, which is where the target directory should generally live. Cargo would normally create that target directory when you run the build command, but it sounds like your environment variable is overriding it to force it to use a target directory in your home directory.

I recommend removing that environment variable; it's unusual and not recommended to always use the same target directory and will probably cause you other hard-to-debug problems like this in the future. If you absolutely have to use that target directory, either use its path directly when calling gdb, or change the target symlinks in each directory in this repository to point to your home directory's target instead. But really I suggest removing that environment variable and letting Cargo do its usual thing of one target directory per project.

connorjcantrell, 2020-05-27:
I removed the environment variable and ran a new build. Adam, I greatly appreciate your help. This was quite an introduction to embedded systems :)

adamgreig, 2020-05-27:
Great, glad to hear you got it working! I updated my previous comment to be a bit clearer that it is supported to use a single target directory, just uncommon. Welcome to embedded Rust!

closed 2020-05-27 by connorjcantrell