rust-embedded/discovery #246: linking with `rust-lld` failed: exit code: 1

imran82ali, 2020-06-12:
Dear Experts,
I am working on Discovery repository, I am building this for STM32F3 Discovery. I am facing following error **linking with `rust-lld` failed: exit code: 1**  When I am building by **cargo build --target thumbv7em-none-eabihf**

It was working fine a couple of weeks ago, I am using Linux 18.04 LTS, I have updated rust my current  version is

imran82ali@gmail:~$ rustup --version --verbose
rustup 1.21.1 (7832b2ebe 2019-12-20)
imran82ali@gmail:~$ rustc --version --verbose
rustc 1.44.0 (49cae5576 2020-06-01)
binary: rustc
commit-hash: 49cae55760da0a43428eba73abcb659bb70cf2e4
commit-date: 2020-06-01
host: x86_64-unknown-linux-gnu
release: 1.44.0
LLVM version: 9.0

Looking forward for your response. 
Thanks, IMRAN ALI

sirhcel, 2020-06-14:
What's the the reason `rust-lld`bails out in your case?

When comming back to [_05-led-roulette_](https://github.com/rust-embedded/discovery/tree/3d4aa95e1a8f5933955168793b5548cd16c27807/src/05-led-roulette) today (which was laying around for several months), I also experienced an issue with `rust-lld`:
```
$ cargo build --target thumbv7em-none-eabihf
[...]
  = note: rust-lld: error: no memory region specified for section '.ARM.exidx'
```
I upated to Rust 1.44 beforehand.

In my case the troublemaker was an outdated _cortex-m-rt_. Updating this crate with `cargo update` to 0.6.12 resolved my issue. The curren version of the [linker fragment `link.x.in` discards the memory section _.ARM.exidx_](https://github.com/rust-embedded/cortex-m-rt/blob/v0.6.12/link.x.in#L155) causing the trouble in my case.

TehseenHasan, 2021-02-01:
I also have the same issue. When I build the **src/05-led-roulette/src/main.rs** code file. It builds fine with no errors. But when I try to build another code in **The Led and Delay abstractions section** of the **Discovery book**. ([Code Link](https://docs.rust-embedded.org/discovery/05-led-roulette/the-led-and-delay-abstractions.html)) that code gives this
 **error: linking with `rust-lld` failed: exit code: 1**
 
I am using Windows 10

The rustup Version is:
rustup 1.23.1 (3df2264a9 2020-11-30)

The rustc Version is:
rustc 1.48.0 (7eac88abb 2020-11-16)

TehseenHasan, 2021-02-01:
Thanks, Sir @imran82ali 
Your solution worked for me.
👇👇👇
Download repo from here: https://github.com/rust-embedded/discovery
then make these changes
open code in VS Code
in **Auxillary** folder there is **cargo.toml** file. there will be **cortex-m** crate, replace that with this:
`cortex-m = "=0.6.4"`

**Note.** add = sign.

babarsaed, 2021-03-02:
Assalam O Alaikum

Dear All, 

I am also facing the same problem when compiling hello-world for itm, i have change in my Auxillary/cargo.toml file (cortex-mm = "=0.6.4" but getting same problem  "linking with `rust-lld` failed: exit code: 1". 
////////////////////////////// Error Message ///////////////////
Compiling stm32f30x-hal v0.2.0
   Compiling f3 v0.6.1
   Compiling aux6 v0.2.0 (/home/babarsaeed/Desktop/06-hello-world/auxiliary)
   Compiling hello-world v0.2.0 (/home/babarsaeed/Desktop/06-hello-world)
error: linking with `rust-lld` failed: exit code: 1

///////////////////////////////////////////////////////////////////////////
My rustup and rustc versions are mentioned below

root@ubuntu:/home/babarsaeed/Desktop/06-hello-world# rustup --version --verbose
rustup 1.23.1 (3df2264a9 2020-11-30)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.49.0 (e1884a8e3 2020-12-29)`

root@ubuntu:/home/babarsaeed/Desktop/06-hello-world# rustc --version --verbose
rustc 1.49.0 (e1884a8e3 2020-12-29)
binary: rustc
commit-hash: e1884a8e3c3e813aada8254edfa120e85bf5ffca
commit-date: 2020-12-29
host: x86_64-unknown-linux-gnu
release: 1.49.0

root@ubuntu:/home/babarsaeed/Desktop/06-hello-world# cargo --version --verbose
cargo 1.49.0 (d00d64df9 2020-12-05)
release: 1.49.0
commit-hash: d00d64df9f803bf5bba8714ca498d8f9159d07f6
commit-date: 2020-12-05

@Sir Imran please help is required for resolving the issue. 

Best Regards

hoani, 2021-04-18:
> **Note.** add = sign.

This is what worked for me, thanks!

sirhcel, 2021-05-31:
Hello everyone, several changes and fixes have landed in this repository recently (for example https://github.com/rust-embedded/discovery/pull/341). They include switching to the actively maintained `stm32f3xx-hal`.

As of https://github.com/rust-embedded/discovery/commit/cc254bbbb381931e22e5f491ad48e558cc5f940d I can build all examples successfully.
```bash
$ git describe --always --dirty
cc254b
$ rustc --version
rustc 1.52.1 (9bc8c42bb 2021-05-09)
$ cargo build --target thumbv7em-none-eabihf 
```
Yes, there are still warnings about unused variables, constants, ... But the build itself works.

I expect this to resolve your issues too. Does the latest updates work out for you?

NitinSaxenait, 2021-06-01:
Hi there, can someone help me with this error?
`
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/Discovery_findings-b7aea3dd29a5e0cc.3ncfi4nyk1i80m6v.rcgu.o" "-o" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/Discovery_findings-b7aea3dd29a5e0cc" "--gc-sections" "-L" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/knoldus/Documents/discovery_findings/target/debug/deps" "-L" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-e65540f12cd59363/out" "-L" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/build/stm32f3-discovery-eeee63eedc8e0380/out" "-L" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/build/cortex-m-4750117c1a38ab5e/out" "-L" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/build/stm32f3-a6b2c85f32baa777/out" "-L" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-d06b3066a590e676.rlib" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/libr0-0d339b2b0521ae60.rlib" "--start-group" "/home/knoldus/Documents/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-68a4462fe5a5c135.rlib" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-ca62b1c0ce05e39c.rlib" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-7be5eea1ddb7319f.rlib" "--end-group" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-09c3e5620e7b76e8.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          

error: aborting due to previous error; 1 warning emitted

error: could not compile `Discovery_findings``

config file
`
[target.thumbv7em-none-eabihf]
#runner = "arm-none-eabi-gdb -q"
runner = "gdb-multiarch -q"
# runner = "gdb -q"
rustflags = [
    "-C", "link-arg=-Tlink.x",
]
[build]
target = "thumbv7em-none-eabihf"`

memory.x is also showing error SOMETHING_IS_GONNA_HAPPEN

sirhcel, 2021-06-01:
Hello @NitinSaxenait, I have no idea what's going wrong at a first glance. Could you please give information on how you got there? Which example are you trying to build? Which revision of the examples and versions of tooling are you using (like I've done in https://github.com/rust-embedded/discovery/issues/246#issuecomment-851465950)?

NitinSaxenait, 2021-06-01:
``@sirhcel i am trying to build the first example from Discovery Book using my own repo.
i am using rustc 1.51.0 

i am getting this error -> error: linking with `rust-lld` failed: exit code: 1
i am not able to link with the microcontroller. I have created the memory.x file but its also giving an error "SOMETHING GONNA HAPPEN"

sirhcel, 2021-06-01:
Alright @NitinSaxenait, but I still have no idea what the error you are getting is about. https://github.com/NitinSaxenait/My_DiscoveryRepo looks pretty empty to me. Could you please publish the actual code that fails and link it here? The compiler looks pretty up to date.

> i am not able to link with the microcontroller. I have created the memory.x file but its also giving an error "SOMETHING GONNA HAPPEN"
Depending on the microcontroller your are using, explicitly providing `memory.x` might not be necessary. Are you writing your code for a STM32F3DISCOVERY?

NitinSaxenait, 2021-06-01:
The code which i want to load in stm32 Discovery board.

`#![no_main]
#![no_std]

use panic_halt as _;
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    let _y;
    let x = 40;
    _y = x;

    // infinite loop; just so we don't leave this stack frame

    loop {}

}
`
on building the code I am getting this error 

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/Discovery_findings-b7aea3dd29a5e0cc.3ncfi4nyk1i80m6v.rcgu.o" "-o" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/Discovery_findings-b7aea3dd29a5e0cc" "--gc-sections" "-L" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/debug/deps" "-L" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-e65540f12cd59363/out" "-L" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/build/stm32f3-discovery-eeee63eedc8e0380/out" "-L" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/build/cortex-m-4750117c1a38ab5e/out" "-L" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/build/stm32f3-a6b2c85f32baa777/out" "-L" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-d06b3066a590e676.rlib" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/libr0-0d339b2b0521ae60.rlib" "--start-group" "/home/knoldus/Documents/My_DiscoveryRepo/discovery_findings/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-68a4462fe5a5c135.rlib" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-ca62b1c0ce05e39c.rlib" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-7be5eea1ddb7319f.rlib" "--end-group" "/home/knoldus/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-09c3e5620e7b76e8.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.

memory.x

MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
}

sirhcel, 2021-06-01:
@NitinSaxenait, I meant publishing the whole project code to a repository and linking it from this issue. The project setup might make the little difference as well ... ;-)

I made the following assumptions about your setup: STM32F3DISCOVERY, directly using `stm32f3xx-hal` because you explicitly mentioned `memory.x`. Under them it seems that the missing interrupt vectors are caused by not actually using the HAL for the MCU which provides them. [Use it in analog to the panic handler](https://github.com/sirhcel/discovery/commit/11d08a1edf3ae0366d273a0f93e552aa29e46f0e) in your `main.rs`. My [test project](https://github.com/sirhcel/discovery/tree/11d08a1edf3ae0366d273a0f93e552aa29e46f0e/src/issue-246-interrupt-vectors) compiles successfully, when having them aboard:
```bash
src/issue-246-interrupt-vectors $ cargo build --target thumbv7em-none-eabihf
[...]
   Compiling interrupt-vectors v0.1.0 ([...]/src/issue-246-interrupt-vectors)
    Finished dev [unoptimized + debuginfo] target(s) in 53.07s
```

NitinSaxenait, 2021-06-02:
@sirhcel that's what i wanted, it worked thanks for your help :)

AthulMuralidhar, 2023-10-14:
> @NitinSaxenait, I meant publishing the whole project code to a repository and linking it from this issue. The project setup might make the little difference as well ... ;-)
> 
> I made the following assumptions about your setup: STM32F3DISCOVERY, directly using `stm32f3xx-hal` because you explicitly mentioned `memory.x`. Under them it seems that the missing interrupt vectors are caused by not actually using the HAL for the MCU which provides them. [Use it in analog to the panic handler](https://github.com/sirhcel/discovery/commit/11d08a1edf3ae0366d273a0f93e552aa29e46f0e) in your `main.rs`. My [test project](https://github.com/sirhcel/discovery/tree/11d08a1edf3ae0366d273a0f93e552aa29e46f0e/src/issue-246-interrupt-vectors) compiles successfully, when having them aboard:
> 
> ```shell
> src/issue-246-interrupt-vectors $ cargo build --target thumbv7em-none-eabihf
> [...]
>    Compiling interrupt-vectors v0.1.0 ([...]/src/issue-246-interrupt-vectors)
>     Finished dev [unoptimized + debuginfo] target(s) in 53.07s
> ```

this totally worked -  thx!