rust-embedded/discovery #253: The book does not get published properly

yerke, 2020-07-03:
As pointed out by @therealprof the book is supposed to be published after merges to `master` branch using https://github.com/rust-embedded/discovery/blob/master/ci/after-success.sh .

Unfortunately, at least from May 2020 the book was not updated on https://docs.rust-embedded.org/discovery/ .

For example, the following 3 PRs are not reflected on the website: https://github.com/rust-embedded/discovery/pull/179, https://github.com/rust-embedded/discovery/pull/148 and https://github.com/rust-embedded/discovery/pull/252 .
https://github.com/rust-embedded/discovery/tree/gh-pages get updated, but not the website itself.

yerke, 2020-07-03:
OK. I think I see what happens. https://docs.rust-embedded.org/discovery/ does not get updated, but https://rust-embedded.github.io/discovery/ does get correctly updated.
https://github.com/rust-embedded/docs is supposed to somehow rebuild 3 books (embedded, discovery, and embedonomicon), but does in practice.

Somewhat related issue: https://github.com/rust-embedded/docs/issues/23

jamesmunns, 2020-07-03:
I think that might be linked to https://github.com/rust-embedded/docs ?

yerke, 2020-07-03:
This issue is a duplicate of https://github.com/rust-embedded/wg/issues/457

yerke, 2020-07-04:
@therealprof Can you please take a look https://github.com/rust-embedded/docs/pull/24 ? Thanks.

yerke, 2020-07-04:
Closed by https://github.com/rust-embedded/docs/pull/24

closed 2020-07-04 by yerke