rust-embedded/discovery #258: 05-led-roulette gdb output on fedora 31 doesn't correspond to examples given in the book and freezes

ragdollfun, 2020-07-29:
Hello,
I'm new to Rust and embedded systems and English is not my first language, so I'm asking for your patience if I don't understand something right away or don't express myself correctly.

I've encountered an issue where the gdb prompt on my system doesn't output the same text as the one present in the book and end up freezing whenever I use the commands "next" or "continue" as instructed in chapter 5.4.

Last details before I show you exactly what output I get: my board is an STM32F3DISCOVERY revision v2 instead of v2.1 like the one used in the book and on fedora 31 neither arm-none-eabi-gdb nor any other package names mentioned in chapter 3.1 seem to be available in fedora's repositories, so I've been using plain gdb.

```
> gdb -q target/thumbv7em-none-eabihf/debug/led-roulette
Reading symbols from target/thumbv7em-none-eabihf/debug/led-roulette...
(gdb) target remote :3333
Remote debugging using :3333
0x00000000 in ?? ()
(gdb) load
Loading section .vector_table, size 0x188 lma 0x8000000
Loading section .text, size 0x3868 lma 0x8000188
Loading section .rodata, size 0x954 lma 0x80039f0
Start address 0x08003766, load size 17220
Transfer rate: 16 KB/sec, 5740 bytes/write.
(gdb) break main
Breakpoint 1 at 0x800018c: file src/05-led-roulette/src/main.rs, line 7.
(gdb) continue
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at src/05-led-roulette/src/main.rs:7
7       #[entry]
(gdb) next
```
And gdb just hangs there and doesn't return anything. Any idea what might be causing this and how to troubleshoot ?

ryankurte, 2020-07-29:
hmm, afaik you will need to run either `arm-none-eabi-gdb` or `gcc-multiarch`, compiled with support for the ARM processor you're using.

it appears that the fedora package build for [arm-none-eabi-gdb](https://koji.fedoraproject.org/koji/packageinfo?packageID=18645) has been broken (and unmaintained) for some time, you may be able to fix this package, or compile [gdb](https://www.gnu.org/software/gdb/) from source with the required target support.

as alternatives you could consider and ubuntu/debian derivative with these tools, or you could try to use docker or podman with one of these distros and a gdb remote server (connecting via TCP), or you could try [eclipse CDT](https://projects.eclipse.org/projects/iot.embed-cdt) which (at least in the past) packaged an arm compiler and GDB instance.

adamgreig, 2020-07-29:
I would have thought you could also just download it as part of the ARM GNU toolchain, which contains pre-built binaries that should run fine on Fedora: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

ryankurte, 2020-07-30:
for some reason i was convinced they didn't ship gdb, have downloaded and checked and it is definitely in there. good call! maybe we should make sure this is front-and-centre in the book.

adamgreig, 2020-07-30:
I absolutely had that in my head as well, got halfway through typing the previous comment, realised they didn't include it, checked the download, found it was in there, finished the comment. In fact I've just realised they also ship `arm-none-eabi-gdb-py` which has the Python extensions, so it works with https://github.com/bnahill/PyCortexMDebug too. Even an old 2018 download had it included, so I have no idea why I thought it wasn't. Anyway we should definitely at least link to it from the book.

ohowland, 2020-07-30:
@ragdollfun I ran into the issue you describe while working through the 05-led-roulette example. I'm on Ubuntu 18.04 and using the gdb-multiarch package.

When proceeding with 'step' instead of 'next', It appears that the program is panicking on `cortex_m::Peripherals::take().unwrap();`
```
cortex_m::peripheral::Peripherals::take () at .../.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.5.10/src/peripheral/mod.rs:158
core::option::Option<T>::unwrap (self=core::option::Option::None) at /rustc/5c1f21c3b82297671ad3ae1e8c942d2ca92e84f2/src/libcore/option.rs:387
```
It looks like this comes from a call in the `aux5::init()` function, located in `05-led-roulette/auxiliary/src/lib.rs`.

I found that updating the `cortex-m` dependency in `05-led-roulette/auxiliary/Cargo.toml` from `0.5.6` to `0.6.3` (current release) fixes the issue. GDB no longer hangs on the 'next' command and the leds are flashing.

```
> ~/dev/discovery/src/05-led-roulette$ cat auxiliary/Cargo.toml 
[package]
authors = ["Jorge Aparicio <jorge@japaric.io>"]
edition = "2018"
name = "aux5"
version = "0.1.0"

[dependencies]
cortex-m = "0.6.3"
panic-halt = "0.2.0"
cortex-m-rt = "0.6.3"

[dependencies.f3]
features = ["rt"]
version = "0.6.1"
```
Hope that helps!

ragdollfun, 2020-07-30:
Thank you everyone for your help. I'll test out your solutions when I have some free time. I'll write back once I have done that.

ragdollfun, 2020-07-30:
It blinks !
...
So updating the `cortex-m` dependency seems to make the board what is expected of it (the LED blinks). Thank you @ohowland !

That said, there is still something strange going on with `gdb`. Following the instructions in chapter 5.4 to the letter doesn't output anything in `gdb` after `next` is called:
```
> gdb -q target/thumbv7em-none-eabihf/debug/led-roulette
Reading symbols from target/thumbv7em-none-eabihf/debug/led-roulette...
(gdb) target remote :3333
Remote debugging using :3333
...
(gdb) load
Loading section .vector_table, size 0x188 lma 0x8000000
Loading section .text, size 0x3838 lma 0x8000188
Loading section .rodata, size 0x8c0 lma 0x80039c0
Start address 0x08003736, load size 17024
Transfer rate: 17 KB/sec, 5674 bytes/write.
(gdb) break main
Breakpoint 1 at 0x800018c: file src/05-led-roulette/src/main.rs, line 7.
(gdb) continue
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at src/05-led-roulette/src/main.rs:7
7       #[entry]
(gdb) next

```
At this point, `gdb` doesn't output anything anymore, but the program is obviously running correctly as the LED on the board blinks. Strangely, a "fix" to this issue is to insert a `step` statement beforehand:
```
(gdb) monitor reset halt
adapter speed: 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003736 msp: 0x10002000
(gdb) load
Loading section .vector_table, size 0x188 lma 0x8000000
Loading section .text, size 0x3838 lma 0x8000188
Loading section .rodata, size 0x8c0 lma 0x80039c0
Start address 0x08003736, load size 17024
Transfer rate: 17 KB/sec, 5674 bytes/write.
(gdb) break main
Note: breakpoint 1 also set at pc 0x800018c.
Breakpoint 2 at 0x800018c: file src/05-led-roulette/src/main.rs, line 7.
(gdb) continue
Continuing.

Breakpoint 1, main () at src/05-led-roulette/src/main.rs:7
7       #[entry]
(gdb) step
led_roulette::__cortex_m_rt_main () at src/05-led-roulette/src/main.rs:9
9           let (mut delay, mut leds): (Delay, Leds) = aux5::init();
(gdb) next
11          let half_period = 500_u16;
(gdb) next
13          loop {
(gdb) next
14              leds[0].on();
(gdb) next
15             delay.delay_ms(half_period);
(gdb) next
17              leds[0].off();
(gdb) next
18              delay.delay_ms(half_period);
(gdb) next
13          loop {
```
But this is not really an issue, so I will try out the GNU Arm Embedded Toolchain from Arm's website when I have more time to troubleshoot. Thank you all again !

closed 2020-07-30 by ragdollfun

therealprof, 2020-07-31:
CC #259

Seems a lot of people are stumbling across that weirdness with the older `cortex-m` these days. 🤔

adamgreig, 2020-07-31:
Yea, I wonder why `take()` is panicking and updating cortex-m fixes it. `cortex-m` v0.5.10 (which isn't tagged in this repo?) brings in a dependency on v0.6.3 anyway; I wonder if something's going wrong in the semver trick when it comes to taking the peripherals or what. We should probably update discovery's cortex-m requirement to 0.6 anyway though.

therealprof, 2020-07-31:
@adamgreig cf. https://github.com/rust-embedded/discovery/issues/259#issuecomment-667007880

adamgreig, 2020-07-31:
I think it's worth investigating, nothing else should be taking them and if something is, updating to the latest version should not fix it, so something's amiss. The whole point of publishing 0.5.10 with the semver trick dependency on 0.6 was to provide backwards compatibility, so if that's somehow not actually working we should find out why.

therealprof, 2020-07-31:
Fair. The example could probably used for experimentation since it is known to fail. Nonetheless we should update the `cortex-m` dependency.