rust-embedded/discovery #261: USART - nothing is printing in minicom (i.e. X)

SuhailArshad1122, 2020-08-11:
Dear Sir,

I tried my best but, no out put show up in minicom, according to this coding (11-USART) X doesn't show up in minicom, please help I am stuck up.

Thanks, Regards
Suhail

SuhailArshad1122, 2020-08-11:
Thanks I got it fixed

closed 2020-08-11 by SuhailArshad1122

fominok, 2020-08-25:
> Thanks I got it fixed

@SuhailArshad1122 can you please share how did you manage to do it? Do you use external serial adapter?

faizsarwar, 2021-02-21:
> Thanks I got it fixed

how did you fixed that i am having same issue please help

vortexofdoom, 2023-06-07:
For any future people who have this problem, I have gone through this twice (having forgotten it) and both times my issue was that I still had it set up for loopback, so disconnecting that worked.