rust-embedded/discovery #263: STM32F3DISCOVERY replacement

bitdivine, 2020-08-24:
Hello all,

Given that the board that he book is based on is no longer available, is it worth making an updated version?  I imagine the easy answer is yes but the next question is what the replacement development board should be.  One from the NUCLEO series?  The STM32F3348-DISCO?

Another question is how to manage multiple copies of the book with different boards.  A branch per board?

Best wishes.

hargoniX, 2020-08-29:
I was gonna look into rewriting large parts of this books as per #262. Without actually having one of these discovery boards it seems kinda hard to me to actually figure out whether my rewrite works though so a big :+1: from my side on using a widely available board being used for this. Until we switch or I get my hands on one I will be kind of blocked on this sadly.

theunkn0wn1, 2020-08-29:
If the discovery board is not obtainable, my suggestion would be to target a nucleo board.
I found the nucleo-64 with the stm32f446retx on it to be easily obtainable. The question then becomes finding a good handful of sensors and the sort to go with.

David-OConnor, 2020-08-29:
Do you have a source on it being NLA? I bought one off Amazon (US) a few days ago.

hargoniX, 2020-08-30:
I talked to another person on Discord yesterday who also reported not being able to find anything online. And just now i went through the first two pages or duck duck go results for the board and couldnt get anything in stock there either, not even at big companies like Mouser.

So even if it might be in stock somewhere it doesnt seem to me like it is trivial to acquire one, hence Id definitely be in for a change.

sebpuetz, 2020-08-31:
Just chiming in here: I tried to get one of these boards (located in Germany) to start the guide and I can't get my hands on any. It's either noted as out-of-stock or delivery date in 4+ weeks.

hargoniX, 2020-09-01:
It has been agreed during the wg meeting that we will be looking into using the micro:bit for the rewrite so stay tuned for that.

Closing as this resolves the original question asked in the issue

closed 2020-09-01 by hargoniX

black-puppydog, 2020-09-06:
@sebpuetz for what it's worth, I was also put off by the 4 week delivery date. ordered it anyhow, thinking I probably wouldn't get to it sooner anyhow, and got it within a few days... :P

bitdivine, 2020-10-11:
Mouser and digi-key have these in stock again.  Perhaps just a supply chain disruption.

shifty331, 2020-12-27:
I think the discovery board is still available. I just bought two discovery boards from a german electronics vendor and they were delivered in two days. As @David-OConnor pointed out, did someone check the official status (as by STMicro)?
Their website says the STM32F3DISCOVERY is active and in volume production.

skaiser, 2021-09-16:
FYI, I just ordered direct from [STMicro's website](https://www.st.com/en/evaluation-tools/stm32f3discovery.html) and received it in 2 days. It was also at least half the price of other websites.

skaiser, 2021-09-16:
And, ironically, it looks like micro:bit may be out of stock as of now, at least in the US.

therealprof, 2021-09-16:
> And, ironically, it looks like micro:bit may be out of stock as of now, at least in the US.

v1 or v2? ;)