rust-embedded/discovery #264: 11-usart None on Peripherals take

fominok, 2020-08-25:
Hi there,
I'm trying to run an example code from `11-usart` section without using external serial module;
Unfortunately, I still cannot get "x" printed in Minicom, also I've noticed that `11-usart/auxiliary/src/lib.rs:18` which is `let cp = cortex_m::Peripherals::take().unwrap();` panics with `None`.

Could you please help me with that?

zhao-lang, 2020-09-01:
getting the same issue. updating `cortex_m` to `0.6.3` stops the panic, but I can't seem to create MonoTimer with the crated updated.

zhao-lang, 2020-09-01:
update - replacing `f3` with `stm32f3xx-hal` more or less got it working, with the caveat that in `main.rs` setting `w.tdr().bits()` now needs to be wrapped in an `unsafe` block. Also the root of the project needed a `memory.x` file.

paulkernfeld, 2020-09-04:
This is related to #259, I believe chapters 11 and 16 are still broken. @zhao-lang I'd love to see a PR with your code. As far as I'm concerned, I'd rather have an `unsafe` solution than a broken one!

black-puppydog, 2020-09-06:
I just ran into this issue and found @zhao-lang's branch. In that branch, the code works **only if** I start with `cargo run --release`. In debug mode, it simply doesn't output anything...

paulkernfeld, 2020-09-07:
Could we just revert to earlier versions of the relevant dependencies? That should be a pretty easy and low-risk way to keep things working if indeed the book is planned to be rewritten for a different board (#263).

bitdivine, 2020-10-11:
@zhao-lang 's code seems correct and runs for me in debug mode.

KernelErr, 2020-12-08:
I bought one board that is produced in June 2020 but still got no output in the terminal while running both the original code and #267 code. However, I could get the loopback message if PC5 and PC4 are connected. That's strange. :(

winksaville, 2021-01-03:
I bought a discovery board in Dec. 2020 and with PR #267 it did work for my board (with the addition of the unsafe work-around). I'd like to second motion to keep the discovery board code up-to-date until a new version of the book is created.

rmokerone, 2021-01-03:
> I bought one board that is produced in June 2020 but still got no output in the terminal while running both the original code and #267 code. However, I could get the loopback message if PC5 and PC4 are connected. That's strange. :(

Has the problem been solved? I have the same problem.

winksaville, 2021-01-03:
It is *not* solved in master. I have a solution in my fork
<https://github.com/winksaville/rust-embedded-discovery> in branch
fix-for-11-usart-using-pr267-and-tweaks
<https://github.com/winksaville/rust-embedded-discovery/tree/fix-for-11-usart-using-pr267-and-tweaks>
.
You can clone and check the branch out that with:

git clone https://github.com/winksaville/rust-embedded-discovery.git
cd rust-embedded-discovery
git checkout fix-for-11-usart-using-pr267-and-tweaks

You should be able to use `cargo run` and see the "X" in the terminal
output:

[image: screenshot-X-after-fix-11-usart.png]

cdparks, 2021-04-10:
I'm seeing the same panic on `master` as of April 10, 2021 using macos Mojave 10.14.6 (18G8012).

hpux735, 2021-04-27:
> I bought one board that is produced in June 2020 but still got no output in the terminal while running both the original code and #267 code. However, I could get the loopback message if PC5 and PC4 are connected. That's strange. :(

I'm having this issue as well.  I've even hooked up my oscilloscope to the pins and I see no activity.  The registers seem to be behaving normally (I've modified it to send the string over and over).  That is, the busy bit is being cleared normally.

adamgreig, 2021-04-27:
Hopefully this issue is (temporarily) fixed by #328, though I'll leave it open until a more long-term fix is in. Thanks @wpd!