rust-embedded/discovery #266: LED Roulette 5.4: set half_period = 100 has no effect

NSchrading, 2020-08-30:
After changing the half_period value to 100 and letting the program continue, the LED continues to blink at the 500ms rate. I can confirm that the half_period value is set to 100, but if I step into the delay_ms call, it reports being called with a value of 500 still:

```
(gdb) info locals
half_period = 100
...
(gdb) step
halted: PC: 0x08002a38
<stm32f30x_hal::delay::Delay as embedded_hal::blocking::delay::DelayMs<u16>>::delay_ms (self=0x10001fa4, ms=500) at \.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f30x-hal-0.2.0\src/delay.rs:38
38              self.delay_ms(u32(ms));
(gdb) info args
self = 0x10001fa4
ms = 500
```

If I end the program, recompile with a value of 100, reload and run, the LED does indeed blink faster as expected.

haussbrandt, 2020-08-30:
Some solutions for this bug can be found in #236

NSchrading, 2020-08-30:
Aha, that makes sense. Thanks and sorry for the duplicate.

closed 2020-08-30 by NSchrading