rust-embedded/discovery #276: Linux install instructions include hacky, somewhat insecure, and often uneeded udev rule which makes mcu world-programmable

tim-seoss, 2020-11-01:
The Linux installation instructions include this sort of thing:

> These rules let you use USB devices like the micro:bit without root privilege, i.e. sudo.
> 
> Create this file in /etc/udev/rules.d with the content shown below.
> 
> $ cat /etc/udev/rules.d/99-microbit.rules
> 
> * CMSIS-DAP for microbit
> SUBSYSTEM=="usb", ATTR{idVendor}=="0d28", ATTR{idProduct}=="0204", MODE:="666"

This is a bit hacky, unnecessary on many Linux distros (e.g. Debian out of the box doesn't need this step I think, since the ACM dev get group writeable by the `plugdev` group) and also opens a bit of a security hole.  Although this is unlikely to be practically exploited, I think it should probably be fixed, since these sort of things have a habit of getting copy-pasted to things like embedded Linux systems, where they might end up doing some damage.

n.b. present on both master and rewrite branches

adamgreig, 2020-11-01:
I totally agree we shouldn't suggest a mode of 666 here; probably we should set the `uaccess` tag and `plugdev` group instead.

However, are you sure it's not required? For one thing, it's not a CDC-ACM device; it's either HID or bulk USB for the CMSIS-DAP interface used to program the board; but even with CDC-ACM I've usually found there does need to be a udev rule if it doesn't already recognise the device. However there are a _lot_ of extra rules hiding on most systems, for example on Ubuntu there's `/lib/udev/rules.d/60-openocd.rules` if you have `openocd` installed, which includes a rule for _all_ CMSIS-DAP devices to put them in the `plugdev` group and apply the `uaccess` tag, along with rules for almost all other probes openocd knows about.

shortspider, 2022-05-11:
I tested with Debian and Ubuntu and they both need rules defined for this to work.

What about setting the owner to the current user (`OWNER:="yourusername"`)?