rust-embedded/discovery #279: Unable to read actual value of I2C_RXDR (Ch-14 I2C)

theinhtut, 2020-12-24:
Hello everyone,
I bought STM32F3DISCOVERY recently and following the examples in Rust Discovery book. I found this book very helpful and interesting to learn for someone like me who is new with Rust.

However, I am stuck at some chapters and one of them is [Chapter-14.4: I2C][Chapter-14.4: I2C]. The value of `I2C_RXDR` is always `0`. I read the reference manual and did a little bit of debugging. `I2C_ISR.RXNE` is updated to `0` from `1` right after `I2C_RXDR` is read. I believe this is the correct behavior according to reference manual _(page 880)_. I am not really sure what causes `I2C_RXDR`'s `0` value. The followings are from itmdump terminal:

Actual output:
``` console
$ # itmdump terminal
0x0A - 0b00000000
```
Expected output: 
``` console
$ # itmdump terminal
0x0A - 0b01001000
```

Same goes for [Chapter-14.5: Reading several registers][Chapter-14.5: I2C]
Actual output:
``` console
$ # itmdump terminal
[0, 0, 0, 0, 0, 0]
[0, 0, 0, 0, 0, 0]
[0, 0, 0, 0, 0, 0]
[0, 0, 0, 0, 0, 0]
(..)
```

Does anyone know what might be the main issue? Or something that I missed out which is not mentioned in the book? This is the board which I have bought from [RS Componens][STM32F3DISCOVERY RS Component].

[image]

[Chapter-14.4: I2C]: https://docs.rust-embedded.org/discovery/14-i2c/the-solution.html
[Chapter-14.5: I2C]: https://docs.rust-embedded.org/discovery/14-i2c/the-solution.html
[STM32F3DISCOVERY RS Component]: https://my.rs-online.com/web/p/microcontroller-development-tools/7692091/?sra=pmpn

wbuck, 2020-12-27:
Go to this [issue](https://github.com/rust-embedded/discovery/issues/274) it'll describe what the issue is.
At the bottom I posted some code you can use instead.

theinhtut, 2020-12-28:
Thank you so much @wbuck 🙏  . Seem like the board that I bought is using `LSM303AGR` instead of `LSM303DLHC`. Closing this for now 😉

closed 2020-12-28 by theinhtut

embear, 2021-07-08:
I have the same problem as @theinhtut . Reading magenetometer delivers `0b00000000` for all registers I try to read. I tried the code @wbuck provided for `LSM303AGR` but that code does not work in general as the `TXIS` bit is never reset. So I think I still have a `LSM303DLHC`
The access to the accelerometer seems to work. The solution code in [16. Punch-o-meter](https://docs.rust-embedded.org/discovery/16-punch-o-meter/index.html) prints meaningful values via ITM.