rust-embedded/discovery #287: error while compiling example 05-led-roulette

bss03arg, 2021-01-26:
cargo build --target thumbv7em-none-eabihf
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.2
   Compiling nb v1.0.0
   Compiling proc-macro2 v1.0.24
   Compiling void v1.0.2
   Compiling cortex-m v0.7.1
   Compiling unicode-xid v0.2.1
   Compiling stable_deref_trait v1.2.0
   Compiling vcell v0.1.3
   Compiling syn v1.0.60
   Compiling cortex-m v0.6.6
   Compiling bitfield v0.13.2
   Compiling cortex-m-rt v0.6.13
   Compiling cortex-m v0.5.10
   Compiling r0 v0.2.2
   Compiling stm32f30x v0.7.1
   Compiling aligned v0.2.0
   Compiling f3 v0.6.1
   Compiling panic-halt v0.2.0
   Compiling nb v0.1.3
   Compiling volatile-register v0.2.0
   Compiling semver v0.9.0
   Compiling generic-array v0.14.4
   Compiling embedded-hal v0.2.4
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling quote v1.0.8
   Compiling generic-array v0.12.3
   Compiling generic-array v0.13.2
   Compiling generic-array v0.11.1
   Compiling l3gd20 v0.2.0
   Compiling as-slice v0.1.4
   Compiling aligned v0.3.4
error[E0432]: unresolved import `cortex_m_0_6::peripheral::itm`
  --> /Users/ccui/.cargo/registry/src/crates.rustcc.com-a21e0f92747beca3/cortex-m-0.5.10/src/peripheral/mod.rs:91:46
   |
91 | pub use cortex_m_0_6::peripheral::{cbp, fpb, itm, tpiu};
   |                                              ^^^ no `itm` in `peripheral`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `cortex-m`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed

I use macbook pro with M1 CPU, rustc 1.50.0-beta.8 (1cd030396 2021-01-20)

adamgreig, 2021-01-26:
Thanks for reporting! For now, you can probably fix this by upgrading `cortex-m` in the `Cargo.toml` file to `0.6` instead of `0.5`; but the underlying problem is a bug in the new 0.6.6 version of cortex-m which will need fixing.

winksaville, 2021-01-26:
@adamgreig suggested the following work-around. He said a "proper" fix should be coming maybe tonight:
```
wink@3900x:~/prgs/rust/tutorial/embedded-discovery/src/05-led-roulette (winks-solutions)
$ git diff .
diff --git a/src/05-led-roulette/auxiliary/Cargo.toml b/src/05-led-roulette/auxiliary/Cargo.toml
index 1b7b80f..9579260 100644
--- a/src/05-led-roulette/auxiliary/Cargo.toml
+++ b/src/05-led-roulette/auxiliary/Cargo.toml
@@ -5,7 +5,7 @@ name = "aux5"
 version = "0.1.0"
 
 [dependencies]
-cortex-m = "0.6.3"
+cortex-m = "=0.6.4"
 panic-halt = "0.2.0"
 cortex-m-rt = "0.6.3"
```
After that change I was able to successfully build:
```
wink@3900x:~/prgs/rust/tutorial/embedded-discovery/src/05-led-roulette (winks-solutions)
$ cargo clean ; cargo build
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.2
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.60
   Compiling stable_deref_trait v1.2.0
   Compiling cortex-m v0.6.4
   Compiling nb v1.0.0
   Compiling vcell v0.1.3
   Compiling cortex-m-rt v0.6.13
   Compiling bitfield v0.13.2
   Compiling cortex-m v0.5.10
   Compiling void v1.0.2
   Compiling stm32f30x v0.7.1
   Compiling aligned v0.2.0
   Compiling r0 v0.2.2
   Compiling f3 v0.6.1
   Compiling panic-halt v0.2.0
   Compiling volatile v0.4.3
   Compiling volatile-register v0.2.0
   Compiling nb v0.1.3
   Compiling semver v0.9.0
   Compiling embedded-hal v0.2.4
   Compiling generic-array v0.14.4
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling quote v1.0.8
   Compiling generic-array v0.12.3
   Compiling generic-array v0.13.2
   Compiling generic-array v0.11.1
   Compiling l3gd20 v0.2.0
   Compiling lsm303dlhc v0.2.0
   Compiling as-slice v0.1.4
   Compiling aligned v0.3.4
   Compiling cortex-m-rt-macros v0.1.8
   Compiling stm32f30x-hal v0.2.0
   Compiling aux5 v0.1.0 (/home/wink/prgs/rust/tutorial/embedded-discovery/src/05-led-roulette/auxiliary)
   Compiling led-roulette v0.1.0 (/home/wink/prgs/rust/tutorial/embedded-discovery/src/05-led-roulette)
    Finished dev [unoptimized + debuginfo] target(s) in 21.30s
```

twitchyliquid64, 2021-01-26:
I'll note we are also hitting this in the ATSAMD CI (ie: https://github.com/atsamd-rs/atsamd/pull/391): https://github.com/atsamd-rs/atsamd/pull/391/checks?check_run_id=1765507423

adamgreig, 2021-01-26:
This is a bug in cortex-m 0.6.6 which we will fix soon, but is there any particular reason you're still using 0.5?

adamgreig, 2021-01-26:
Fixed by the new 0.6.7 release. Thanks again for reporting!

closed 2021-01-26 by adamgreig

winksaville, 2021-01-26:
Verified it works for me, txs @adamgreig !

bss03arg, 2021-01-27:
HI all,

The problem was fixed with `cortex-m = "=0.6.7"` in auxiliary/Cargo.toml .
It seems that there's still some other package using cortex-m v0.5.10. But the project works, it's ok now.
thank you very much. 

cargo readobj --target thumbv7em-none-eabihf --bin led-roulette -- -file-headers
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.2
   Compiling nb v1.0.0
   Compiling void v1.0.2
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling cortex-m v0.7.1
   Compiling vcell v0.1.3
   Compiling syn v1.0.60
   Compiling stable_deref_trait v1.2.0
   Compiling cortex-m v0.6.7
   Compiling bitfield v0.13.2
   Compiling cortex-m v0.5.10
   Compiling cortex-m-rt v0.6.13
   Compiling stm32f30x v0.7.1
   Compiling aligned v0.2.0
   Compiling r0 v0.2.2
   Compiling f3 v0.6.1
   Compiling panic-halt v0.2.0
   Compiling nb v0.1.3
   Compiling volatile-register v0.2.0
   Compiling semver v0.9.0
   Compiling generic-array v0.14.4
   Compiling embedded-hal v0.2.4
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling quote v1.0.8
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling generic-array v0.11.1
   Compiling l3gd20 v0.2.0
   Compiling as-slice v0.1.4
   Compiling aligned v0.3.4
   Compiling lsm303dlhc v0.2.0
   Compiling cortex-m-rt-macros v0.1.8
   Compiling stm32f30x-hal v0.2.0
   Compiling aux5 v0.1.0 (/Users/ccui/CLionProjects/discovery/src/05-led-roulette/auxiliary)
   Compiling led-roulette v0.1.0 (/Users/ccui/CLionProjects/discovery/src/05-led-roulette)
    Finished dev [unoptimized + debuginfo] target(s) in 19.20s
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x8000189
  Start of program headers:          52 (bytes into file)
  Start of section headers:          884916 (bytes into file)
  Flags:                             0x5000400
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         4
  Size of section headers:           40 (bytes)
  Number of section headers:         21
  Section header string table index: 19