rust-embedded/discovery #288: 05-led-roulette: "The led and delay abstractions" code snippet causes "linking with rust-lld failed: exit code 1"

ccod, 2021-01-28:
I was able to run the previous segments of the lesson, but couldn't compile this new bit of code:<br/>
` let (mut delay, mut leds): (Delay, Leds) = aux5::init(); `

if this helps: `rustup --version --verbose`<br/>
```bash
rustup 1.23.1 (3df2264a9 2020-11-30)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.49.0 (e1884a8e3 2020-12-29)`
```

Some of the notes for the error: 
```bash
error: linking with `rust-lld` failed: exit code: 1
[...]
 = note: rust-lld: error: duplicate symbol: __bkpt
         rust-lld: error: duplicate symbol: __cpsid
         rust-lld: error: duplicate symbol: __cpsie
         rust-lld: error: duplicate symbol: __delay
         rust-lld: error: duplicate symbol: __dmb
         rust-lld: error: duplicate symbol: __dsb
         rust-lld: error: duplicate symbol: __isb
         rust-lld: error: duplicate symbol: __msp_r
         rust-lld: error: duplicate symbol: __msp_w
         rust-lld: error: duplicate symbol: __nop
         rust-lld: error: duplicate symbol: __psp_r
         rust-lld: error: duplicate symbol: __psp_w
         rust-lld: error: duplicate symbol: __sev
         rust-lld: error: duplicate symbol: __wfe
         rust-lld: error: duplicate symbol: __wfi
```

If there is more information to require to address the issue, let me know.

adamgreig, 2021-01-28:
Huh, this is a new one, thanks for reporting!

There might be more information in the build output, could you possibly paste the full output instead? Another useful diagnostic would be the output of `cargo tree` to see what versions of dependencies are being used. You might try running `cargo update` just in case something needs updating (there have been some recent releases of the cortex-m crate which are probably the underlying problem).

It would also be helpful to know exactly what command you're running and in what directory. Thanks!

phijor, 2021-01-28:
I just ran into exactly the same problem.

Steps to reproduce:

1. Clone a fresh copy:
    ```sh
    cd /tmp && git clone https://github.com/rust-embedded/discovery.git && cd $_
    ```
2. Check-out the (now) current `master`:
    ```
    git checkout 86fde2b
    ```
3. Apply the following changes:
    ```diff
    diff --git a/src/05-led-roulette/src/main.rs b/src/05-led-roulette/src/main.rs
    index 26e4499..7cd11ce 100644
    --- a/src/05-led-roulette/src/main.rs
    +++ b/src/05-led-roulette/src/main.rs
    @@ -6,9 +6,7 @@ use aux5::entry;
    
    #[entry]
    fn main() -> ! {
    -    let _y;
    -    let x = 42;
    -    _y = x;
    +    let (_leds, _delay) = aux5::init();
    
        // infinite loop; just so we don't leave this stack frame
        loop {}
    ```
4. Build (`cd src/05-led-roulette && cargo build --target thumbv7em-none-eabihf`), observe the following output:
    <details>
        <summary>Click to expand</summary>

    ```
    Compiling led-roulette v0.1.0 (/tmp/discovery/src/05-led-roulette)
        error: linking with `rust-lld` failed: exit code: 1
        |
        = note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" "/home/phijor/etc/data/rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/led_roulette-0f2a68f050c96529.3g0s4l3mjwgx5oz5.rcgu.o" "-o" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/led_roulette-0f2a68f050c96529" "--gc-sections" "-L" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps" "-L" "/tmp/discovery/target/debug/deps" "-L" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-3fab0cdaabbc0f2d/out" "-L" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-74d36e1f1aabd6ca/out" "-L" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/build/f3-a7b915113a3d824f/out" "-L" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-6161e248155f0419/out" "-L" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/build/stm32f30x-e0416ac8c5c0b4ff/out" "-L" "/home/phijor/etc/data/rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libaux5-2647278883ec76f8.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libf3-b31e3c906f998103.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x_hal-96f7e9b797758484.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-2f74b3226f113d80.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-39fa569373b8745d.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-713ce8358560bc03.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-877a76d5b7010d65.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libaligned-02103f11180580f7.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libas_slice-de7eb69ac0147d18.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-32b2172bdb0028d2.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-a4597f99ceadd11b.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-bb9dfc47d21e0ab7.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-25666554ec7c606b.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-e9dfc9a66e182937.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libvcell-460cebbd3d28a906.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-26dd7c8407e43aca.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libaligned-a5f6002334f853cd.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/liblsm303dlhc-fee15414b625f1a3.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcast-10bf1c29315a4ab8.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libl3gd20-4f3b7e22208673f8.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-dc7b5897df6ccaf3.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libtypenum-8a1332b3629216f3.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-1976ddfb65d07733.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libvoid-4476849701c0ad80.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libnb-868cf5f4eaf62c12.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libnb-64a642215922cc8e.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-da43fa24ed3aaa7c.rlib" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libr0-06e6ec7a6680e345.rlib" "--start-group" "/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-e385674ff70fc036.rlib" "/home/phijor/etc/data/rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-2861a034b7edfae2.rlib" "/home/phijor/etc/data/rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-93e0ebeeb48255f4.rlib" "--end-group" "/home/phijor/etc/data/rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-d6eaad3e7b816578.rlib" "-Tlink.x" "-Bdynamic"
        = note: rust-lld: error: duplicate symbol: __bkpt
[... cut ...]
    │       ├── nb v0.1.3 (*)
    │       ├── stm32f30x v0.7.1
    │       │   ├── bare-metal v0.2.5 (*)
    │       │   ├── cortex-m v0.5.10 (*)
    │       │   ├── cortex-m-rt v0.6.13 (*)
    │       │   └── vcell v0.1.3
    │       └── void v1.0.2
    └── panic-halt v0.2.0
```

</details>

Rust verion:

    rustup 1.23.1 (2020-12-18)
    info: This is the version for the rustup toolchain manager, not the rustc compiler.
    info: The currently active `rustc` version is `rustc 1.49.0 (e1884a8e3 2020-12-29)`

ccod, 2021-01-28:
I ran `cargo build --target thumbv7em-none-eabihf` in the `discovery/src/05-led-roulette` directory.

The full output from the build command:
```bash
   Compiling led-roulette v0.1.0 (/home/ccod/Code/Testing/discovery/src/05-led-roulette)
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" "/home/ccod/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/led_roulette-0f2a68f050c96529.3g0s4l3mjwgx5oz5.rcgu.o" "-o" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/led_roulette-0f2a68f050c96529" "--gc-sections" "-L" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/ccod/Code/Testing/discovery/target/debug/deps" "-L" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-3fab0cdaabbc0f2d/out" "-L" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-74d36e1f1aabd6ca/out" "-L" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/build/f3-a7b915113a3d824f/out" "-L" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/build/cortex-m-6161e248155f0419/out" "-L" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/build/stm32f30x-e0416ac8c5c0b4ff/out" "-L" "/home/ccod/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libaux5-2647278883ec76f8.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libf3-b31e3c906f998103.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x_hal-96f7e9b797758484.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-2f74b3226f113d80.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-39fa569373b8745d.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-713ce8358560bc03.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-877a76d5b7010d65.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libaligned-02103f11180580f7.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libas_slice-de7eb69ac0147d18.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-32b2172bdb0028d2.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-a4597f99ceadd11b.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-bb9dfc47d21e0ab7.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-25666554ec7c606b.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-e9dfc9a66e182937.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libvcell-460cebbd3d28a906.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-26dd7c8407e43aca.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libaligned-a5f6002334f853cd.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/liblsm303dlhc-fee15414b625f1a3.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libcast-10bf1c29315a4ab8.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libl3gd20-4f3b7e22208673f8.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-dc7b5897df6ccaf3.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libtypenum-8a1332b3629216f3.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-1976ddfb65d07733.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libvoid-4476849701c0ad80.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libnb-868cf5f4eaf62c12.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libnb-64a642215922cc8e.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-da43fa24ed3aaa7c.rlib" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libr0-06e6ec7a6680e345.rlib" "--start-group" "/home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-e385674ff70fc036.rlib" "/home/ccod/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-2861a034b7edfae2.rlib" "/home/ccod/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-93e0ebeeb48255f4.rlib" "--end-group" "/home/ccod/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-d6eaad3e7b816578.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: duplicate symbol: __bkpt
          >>> defined at cortex-m.o:(.text.__bkpt+0x1) in archive /home/ccod/Code/Testing/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-39fa569373b8745d.rlib
          >>> defined at inline.rs:13 (asm/inline.rs:13)
[... cut ...]
    │       │   ├── aligned v0.2.0
    │       │   ├── bare-metal v0.2.5 (*)
    │       │   ├── cortex-m v0.6.7 (*)
    │       │   └── volatile-register v0.2.0 (*)
    │       ├── embedded-hal v0.2.4 (*)
    │       ├── nb v0.1.3 (*)
    │       ├── stm32f30x v0.7.1
    │       │   ├── bare-metal v0.2.5 (*)
    │       │   ├── cortex-m v0.5.10 (*)
    │       │   ├── cortex-m-rt v0.6.13 (*)
    │       │   └── vcell v0.1.3
    │       └── void v1.0.2
    └── panic-halt v0.2.0
```

Hope that helps

adamgreig, 2021-01-29:
Thanks for the very detailed reproduction! I'm still investigating exactly what's triggering this, but in the meantime you can work around it by editing auxiliary/Cargo.toml and changing the version of cortex-m to `=0.6.4` (including the `=`):

```toml
[dependencies]
cortex-m = "=0.6.4"
```

imran82ali, 2021-01-29:
There are number of developers who are facing the same issue. Error Resolved

adamgreig, 2021-02-01:
This is now fixed by 0.5.11, although you might need to run "cargo update" on an already checked-out repository. Fresh checkouts (or any without a Cargo.lock) should work immediately. Sorry for the wait and thanks for reporting the bug!

closed 2021-02-01 by adamgreig