rust-embedded/discovery #295: "The Led and Delay abstractions" chapter code doesn't work

SergeiGaliullin, 2021-02-05:
Hi, I'm following the book and on **The Led and Delay abstractions** there is following code:
```
#![deny(unsafe_code)]
#![no_main]
#![no_std]

use aux5::{entry, prelude::*, Delay, Leds};

#[entry]
fn main() -> ! {
    let (mut delay, mut leds): (Delay, Leds) = aux5::init();

    let half_period = 500_u16;

    loop {
        leds[0].on();
        delay.delay_ms(half_period);

        leds[0].off();
        delay.delay_ms(half_period);
    }
}
```
It doesn't run. 
```
~/.../src/05-led-roulette >>> cargo build --target thumbv7em-none-eabihf                                                  ±[●][master]
   Compiling aux5 v0.2.0 (/home/serge/Projects/Study/Rust/src/05-led-roulette/auxiliary)
   Compiling led-roulette v0.2.0 (/home/serge/Projects/Study/Rust/src/05-led-roulette)
error[E0432]: unresolved import aux5::prelude
 --> src/05-led-roulette/src/main.rs:5:19
  |
5 | use aux5::{entry, prelude::*, Delay, Leds};
  |                   ^^^^^^^ could not find prelude in aux5

error[E0308]: mismatched types
 --> src/05-led-roulette/src/main.rs:9:48
  |
9 |     let (mut delay, mut leds): (Delay, Leds) = aux5::init();
  |                                -------------   ^^^^^^^^^^^^ expected struct `Leds`, found array of 8 elements
  |                                |
  |                                expected due to this
  |
  = note: expected tuple `(Delay, Leds)`
             found tuple `(Delay, [Switch<PEx<Output<PushPull>>, ActiveHigh>; 8])`

error[E0608]: cannot index into a value of type `Leds`
  --> src/05-led-roulette/src/main.rs:14:9
   |
14 |         leds[0].on();
   |         ^^^^^^^

error[E0599]: no method named `delay_ms` found for struct `Delay` in the current scope
  --> src/05-led-roulette/src/main.rs:15:15
   |
15 |         delay.delay_ms(half_period);
   |               ^^^^^^^^ method not found in `Delay`
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
5  | use aux5::DelayMs;
   |

error[E0608]: cannot index into a value of type `Leds`
  --> src/05-led-roulette/src/main.rs:17:9
   |
17 |         leds[0].off();
   |         ^^^^^^^

error[E0599]: no method named `delay_ms` found for struct `Delay` in the current scope
  --> src/05-led-roulette/src/main.rs:18:15
   |
18 |         delay.delay_ms(half_period);
   |               ^^^^^^^^ method not found in `Delay`
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
5  | use aux5::DelayMs;
   |

error: aborting due to 6 previous errors

Some errors have detailed explanations: E0308, E0432, E0599, E0608.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `led-roulette`

To learn more, run the command again with --verbose.
```

So I looked into **auxiliary** folder and indeed didn't find these items. Then I looked through github history of the book and attempted to use older versions of these files, but that didn't work either. Thanks for helping in advance.

shaukathussain, 2021-02-06:
I have copied the repository on 24 January that is working fine, but recently someone messed up this repository two days ago. it wasted my whole day troubleshooting the problem. then I found the old file somehow and worked with that one..

pevandenburie, 2021-02-07:
Duplicate of #280

adamgreig, 2021-02-07:
Why do you think it's a duplicate? I'm not sure, I think this issue is because the published book hasn't been updated because our CI is broken, and so the code in the repository is now out of date compared to the code snippets on the website - e.g. the repository's aux5::init method now returns types (Delay, LedArray) but the published book snippet expects (Delay, Leds).

#280 looks more like a change to the compiler optimisation meaning the gdb interactive stuff isn't working right, but it predates the changes to chapter 5 that are causing the issues here.

pevandenburie, 2021-02-07:
Sorry you are right, I mixed up issues.

Robotronics, 2021-02-18:
> Hi, I'm following the book and on **The Led and Delay abstractions** there is following code:
>
> ```
> #![deny(unsafe_code)]
> #![no_main]
> #![no_std]
>
> use aux5::{entry, prelude::*, Delay, Leds};
>
> #[entry] [...]

Hi, I had the same problem. It appears the book's example is out of date however I discovered that the correct content is in the "05-led-roulette" directory in a file called "the-led-and-delay-abstractions.md". 
Cheers,
Dane.

GreenPenguino, 2021-02-20:
You can serve the book on localhost to make sure you have the most up to date version:
- run `cargo install mdbook`
- run `mdbook build`
- run `mdbook serve` (leave this terminal open)
- open `localhost:3000` in the browser

adamgreig, 2021-02-21:
The hosted version here should always be up to date too, it's built after each commit:

https://rust-embedded.github.io/discovery (linked to from this repo's info, too).

Unfortunately our _other_ hosted copy is not auto-updating properly, which we're hoping to fix soon:

https://docs.rust-embedded.org/discovery

spartacoos, 2021-06-12:
Hi, I'm going through the discovery embedded book and ran into a strange issue that might be related to this. For example, the time works fine most of the time. I.e if I set half_period = 500_u16 it blinks for what seems like half a second, etc. However, if I start to make that value bigger it then fails. 

Steps:
$ cargo run
`(gdb) target remote :3333` 
`(gdb) load`
`(gdb) b main.rs:15` (beginning of loop)
`(gdb) continue`
--stops at loop--
`(gdb) disable 1`
`(gdb) continue`
--blinks led (as expected) --

However if I exit this gdb session and change the main.rs so that `half_period = 5000_u16;` (10x bigger) or some other large values then sometimes it doesn't work and sometimes it does work. I've spent the last couple hours trying to figure out if this is a gdb issue or something else but it doesn't make much sense to me. 

Steps:
$ cargo run
`(gdb) target remote :3333` 
`(gdb) load`
`(gdb) b main.rs:15` (beginning of loop)
`(gdb) continue`
--hangs--

For further context here is the full main.rs and the [link](https://docs.rust-embedded.org/discovery/05-led-roulette/the-led-and-delay-abstractions.html) to the page in the book.

```
#![deny(unsafe_code)]
#![no_main]
#![no_std]

use volatile::Volatile;
use aux5::{entry, Delay, DelayMs, LedArray, OutputSwitch};

#[entry]
fn main() -> ! {
    let (mut delay, mut leds): (Delay, LedArray) = aux5::init();

    let mut half_period = 2048_u16;//2500_u16;
    let v_half_period = Volatile::new(&mut half_period);

    loop {
        leds[0].on().ok();
        delay.delay_ms(v_half_period.read());

        leds[0].off().ok();
        delay.delay_ms(v_half_period.read());
    }
}

```

skaiser, 2021-09-13:
> You can serve the book on localhost to make sure you have the most up to date version:
> 
> * open a terminal in the repository
> * run `cargo install mdbook`
> * run `mdbook build`
> * run `mdbook serve` (leave this terminal open)
> * open `localhost:3000` in the browser

$ mdbook build
2021-09-13 16:30:51 [INFO] (mdbook::book): Book building has started
2021-09-13 16:30:51 [INFO] (mdbook::book): Running the html backend
2021-09-13 16:30:52 [ERROR] (mdbook::utils): Error: Rendering failed
2021-09-13 16:30:52 [ERROR] (mdbook::utils): 	Caused By: Failed to read "/Users/skaiser/code/opensource/discovery/src/07-registers/target"
2021-09-13 16:30:52 [ERROR] (mdbook::utils): 	Caused By: No such file or directory (os error 2)

I had to cd into src/07-registers and run cargo build, then mdbook build and serve worked.